NordicSemiconductor/IOS-DFU-Library #2: Integration documentation

starblanket, 2016-03-10:
There are steps missing in the documentation on how to integrate the library. I'm not sure if this is happening only because I'm using the library from a non-swift project, but under the target's Build Options, I had to set  "Embedded Content Contains Swift Code" to YES. Without this option set, the program will link but fail on the target with:

     dyld: Library not loaded: @rpath/libswiftCore.dylib ...

starblanket, 2016-03-10:
Never mind! I see that step was there all along...

closed 2016-03-10 by starblanket

philips77, 2016-03-11:
Hi, actually it looks like this option should be checked for the target, and unchecked for the framework's target. Otherwise xcode adds those libs twice and AppStore rejects the app. I still struggle with a proper configuration for AppStore, last time I got Build Invalid.