NordicSemiconductor/IOS-DFU-Library #3: DFUServiceDelegate does not receive a StateCompleted event, although peripheral flashes fine

starblanket, 2016-03-12:
The DFUServiceDelegate never receives a StateCompleted event with version 1.0 of the library. During a flashing update, the delegate receives these state changes: StateConnecting, StateStarting, StateEnablingDfuMode, StateStarting, StateUploading, StateValidating, StateDisconnecting. After receiving the StateDisconnecting event, I expected a StateCompleted event but instead the delegate is informed there was an error writing the "activate and reset" to the control point characteristic. It's not a legitimate error to report to the delegate though because the peripheral actually did reset.

```
-[OTAViewController didStateChangedTo:]:121 DFU state change to 4
-[OTAViewController logWith:message:]:141 Data written to 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:141 Validate Firmware (Op Code = 4) request sent
-[OTAViewController logWith:message:]:141 Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100401
-[OTAViewController logWith:message:]:141 Response (Op Code = 4, Status = 1) received
-[OTAViewController logWith:message:]:141 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:141 peripheral.writeValue(0x05, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
-[OTAViewController didStateChangedTo:]:121 DFU state change to 5
-[OTAViewController logWith:message:]:141 Data written to 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:141 Activate and Reset (Op Code = 5) request sent
-[OTAViewController logWith:message:]:141 [Callback] Central Manager did disconnect peripheral
-[OTAViewController logWith:message:]:141 Disconnected by the remote device
-[OTAViewController didErrorOccur:withMessage:]:126 error 305 Writing to characteristic failed
```

I just want to add that this is great library -- it's made it so simple and the flashing is working reliably!

philips77, 2016-03-12:
Hi, what is your bootloader, DFU and SD versions? While testing I didn't get this "error", which of course, as you say, is not an error. The device got reset before sending ack.

philips77, 2016-03-12:
In DFUControlPoint class, after sending the Activate and Reset command, looks like you get an error but the 2 logs are not shown on your listing. Looks like you intentionally skip logs with error level as others seems to be written.
I'll add a flag to this class, set during sending Reset of AoR commands. Then, any error should be ignored in didCharacteristicWrite. I'll try it on Monday.

starblanket, 2016-03-12:
The logs above are a record of everything that came to `[LoggerDelegate logWith:message:]`. All log levels were output (the `level` parameter was just ignored).

The peripheral's bootloader+app+SD was built using the most up to date code from http://mbed.org/teams/Nordic-Semiconductor/code/nRF51822, which shows a version of "Nordic BLE4.1 ver:8 fw:0067" from `BLE::getVersion()`. I have trouble correlating this with anything on the Nordic site. From the documentation I've only seen that Soft Device S110 8.0.0 should have a fw id of 0x0064. If you know of a better way to find what version of the soft device is flashed, let me know.

philips77, 2016-03-14:
Hi, according to this page: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk51.v9.0.0/bledfu_example_init.html?cp=4_0_4_4_3_1_3 your SoftDevice version is S130, version 1.0.
I'm updating the DFU Library just now. Hope it will fix your issue.

philips77, 2016-03-14:
Hi, I did push some changes to nRF Toolbox and the DFU Library a second ago. I still have no idea how you got those errors, as from the logs I see you got didWriteValueForCharacteristic without an error (it has logged:
```
-[OTAViewController logWith:message:]:141 Data written to 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:141 Activate and Reset (Op Code = 5) request sent
```
but also it called didErrorOccur. But at least now it should not throw an error if writing Activate and Reset "failed" to be sent.
If it doesn't work, could you check and suggest a change? I'm leaving for vacation tomorrow, so you're on your own for some time :)

starblanket, 2016-03-15:
I don't know swift, but it looks like the `DFUExecutor` is caching an error, and reports it whenever `didDeviceDisconnect` is called. `didDeviceDisconnect` is invoked when the transfer completes. Once I saw that I examined the log more closely. The error was not from the last write that was performed, but from much earlier in the flashing, when enabling DFU mode. It's **_that_** error that should not get propagated to the delegate at the end of the transfer because the flashing was actually successful.

Here is the original error in the log (with logging levels included now in the output):

```
-[OTAViewController didStateChangedTo:]:121 DFU state change to 1
-[OTAViewController logWith:message:]:141 01 Notifications enabled for 00001534-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:141 10 DFU Control Point notifications enabled
-[OTAViewController logWith:message:]:141 15 Application with buttonless update found
-[OTAViewController logWith:message:]:141 01 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:141 00 peripheral.writeValue(0x0104, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
-[OTAViewController didStateChangedTo:]:121 DFU state change to 2
-[OTAViewController logWith:message:]:141 20 Writing to characteristic failed
-[OTAViewController logWith:message:]:141 20 Error 14: Unlikely error.
-[OTAViewController logWith:message:]:141 01 Disconnecting...
```

I confirmed it was actually `CBATTErrorUnlikelyError` being reported at the end of the flashing by beefing up the logger output in `DFUControlPoint::peripheral: didWriteValueForCharacteristic: error:` to include the actual error code (this might be a useful addition to the code base).

philips77, 2016-03-15:
Hi, that's very good :)
Unfortunately I started vacations today so I won't be able to do the fix soon. But you can easily do this. 
Yesterday I pushed some changes. You just need to add the .JumpToBootloader request to the switch in DFUControlPoint's send(request:onSuccess:onError) method, next to .ActivateAndReset and .Reset. This command also causes a disconnection and reset, I forgot to add it. And modify the comment a bit in this 'case'.
Could you do a push request?

philips77, 2016-03-15:
But the true is, that if the error occurs before switching to the bootloader mode (like in your case), or perhaps while sending part 1 if there are 2 parts, the error is reported at the very end instead of throwing it after the soonest disconnect.
I'll look at it when I'm back.

philips77, 2016-03-16:
Hi, I've just added a commit that should fix your issue. Could you please verify it? I did the change from the web interface.

starblanket, 2016-03-19:
It looks good now. Thanks!

```
-[OTAViewController didStateChangedTo:]:121 DFU state change to 1
-[OTAViewController logWith:message:]:141 01 Notifications enabled for 00001534-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:141 10 DFU Control Point notifications enabled
-[OTAViewController logWith:message:]:141 15 Application with buttonless update found
-[OTAViewController logWith:message:]:141 01 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:141 00 peripheral.writeValue(0x0104, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
-[OTAViewController didStateChangedTo:]:121 DFU state change to 2
-[OTAViewController logWith:message:]:141 10 Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
-[OTAViewController logWith:message:]:141 15 Device disconnected before sending ACK
-[OTAViewController logWith:message:]:141 15 Error 14: Unlikely error.
-[OTAViewController logWith:message:]:141 00 [Callback] Central Manager did disconnect peripheral
-[OTAViewController logWith:message:]:141 05 Disconnected by the remote device
...
-[OTAViewController didStateChangedTo:]:121 DFU state change to 5
-[OTAViewController logWith:message:]:141 05 Data written to 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:141 10 Activate and Reset (Op Code = 5) request sent
-[OTAViewController logWith:message:]:141 00 [Callback] Central Manager did disconnect peripheral
-[OTAViewController logWith:message:]:141 05 Disconnected by the remote device
-[OTAViewController didStateChangedTo:]:121 DFU state change to 6
```

closed 2016-03-19 by starblanket