NordicSemiconductor/IOS-DFU-Library #4: missing required architecture x86_64 for file DFULibrary.framework/DFULibrary (2 slices)

zubairbh, 2016-03-15:
building a Cordova/PhoneGap Plugin (kind of a wrapper around the library so we can use it in phonegap Apps) but i get the error "missing required architecture x86_64"

when i see build settings for the DFULibrary it has included "arm64"  in its builds .

please see complete log from xcode  below

`Ld /Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Products/Debug-iphonesimulator/HelloCordova.app/HelloCordova normal x86_64
    cd /Users/applezone/Desktop/waqar/ZipFiles/NewShortPluginandProject/again/platforms/ios
    export IPHONEOS_DEPLOYMENT_TARGET=8.0
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator9.2.sdk -L/Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Products/Debug-iphonesimulator -L/Users/applezone/Desktop/waqar/ZipFiles/NewShortPluginandProject/again/platforms/ios/HelloCordova -L/Users/applezone/Desktop/waqar/ZipFiles/NewShortPluginandProject/again/platforms/ios -F/Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Products/Debug-iphonesimulator -F/Users/applezone/Desktop/waqar/ZipFiles/NewShortPluginandProject/again/platforms/ios -FHelloCordova/Plugins/com.example.hello -filelist /Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Intermediates/HelloCordova.build/Debug-iphonesimulator/HelloCordova.build/Objects-normal/x86_64/HelloCordova.LinkFileList -Xlinker -rpath -Xlinker @executable_path/Frameworks -mios-simulator-version-min=8.0 -Xlinker -objc_abi_version -Xlinker 2 -ObjC -fobjc-arc -fobjc-link-runtime -lz.1.1.3 /Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Products/Debug-iphonesimulator/libCordova.a -lCorePlot-CocoaTouch -framework DFULibrary -Xlinker -dependency_info -Xlinker /Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Intermediates/HelloCordova.build/Debug-iphonesimulator/HelloCordova.build/Objects-normal/x86_64/HelloCordova_dependency_info.dat -o /Users/applezone/Library/Developer/Xcode/DerivedData/HelloCordova-dttpzwwhzrracuapnpqjorsjviwa/Build/Products/Debug-iphonesimulator/HelloCordova.app/HelloCordova

ld: warning: ignoring file HelloCordova/Plugins/com.example.hello/DFULibrary.framework/DFULibrary, missing required architecture x86_64 in file HelloCordova/Plugins/com.example.hello/DFULibrary.framework/DFULibrary (2 slices)
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$__TtC10DFULibrary11DFUFirmware", referenced from:
      objc-class-ref in DFUplugin.o
  "_OBJC_CLASS_$__TtC10DFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in DFUplugin.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

ld: warning: ignoring file HelloCordova/Plugins/com.example.hello/DFULibrary.framework/DFULibrary, missing required architecture x86_64 in file HelloCordova/Plugins/com.example.hello/DFULibrary.framework/DFULibrary (2 slices)

  "_OBJC_CLASS_$__TtC10DFULibrary11DFUFirmware", referenced from:
      objc-class-ref in DFUplugin.o
  "_OBJC_CLASS_$__TtC10DFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in DFUplugin.o

ld: symbol(s) not found for architecture x86_64

clang: error: linker command failed with exit code 1 (use -v to see invocation)`

philips77, 2016-03-16:
Hi,
could you change the project settings and do a push request?

zubairbh, 2016-03-16:
let me see .
also i think this problem is related if the project is being run on a simulator .the architecture x86_64 is i think needed for the iOS simulator as it is running on the PC.
when i tried with actual device it worked.

y2kpaulh, 2016-03-22:
Same problem exist. but, I can't find solution in simulator operation condition.

In my case,  actual device it worked, too.
but, iOS simulator condition is failed.

if I add x86_64 in library architecture,  success  compile in iOS simulator.
but, Actual device condition not worked.

klemenzagar91, 2016-04-07:
Hi, I have similar problem. When I run it on simulator I get this warnings 
ld: warning: ignoring file /Users/Klemen/Dev/Scoutee/ScouteeGit/Xcode/Frameworks/DFULibrary.framework/DFULibrary, missing required architecture i386 in file /Users/Klemen/Dev/Scoutee/ScouteeGit/Xcode/Frameworks/DFULibrary.framework/DFULibrary (2 slices)
Undefined symbols for architecture i386:
  "_OBJC_CLASS_$__TtC10DFULibrary11DFUFirmware", referenced from:
      objc-class-ref in ScouteeManager.o
  "_OBJC_CLASS_$__TtC10DFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in ScouteeManager.o
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)

klemenzagar91, 2016-04-07:
How can I fix this.

philips77, 2016-04-07:
Hi, I'm not really pro in xcode projects configuration.. I guess you are all using the framework from a release zip. It was not built for i368 arch, as I'm not using emulator :)
But I guess you should be able to include the project to your workspace, link the product as embedded framework and set the build settings yourself.
I will come back to you here, I need to check it, perhaps release a new release build for all architectures.

klemenzagar91, 2016-04-07:
Yeah, Im using zip release. I did build DFULibrary project and than copy the framework from product folder into my project.

klemenzagar91, 2016-04-07:
Do I need to update build setting in my xcode project or in DFULibrary project and then again import framework in my project?

philips77, 2016-04-07:
In the DFU Lib project build settings you have architectures. I guess you have to add i386 there and compile again? I'd have to read about it.

klemenzagar91, 2016-04-07:
Ok, I'll also try to find more information about this.

klemenzagar91, 2016-04-07:
Hi. The warning goes away when I set Build active architecure only to NO.

mostafaberg, 2016-04-14:
Hi !, I'm looking into this issue, we're trying to bring you a fat library with all the archs, but it seems like some bug is in Xcode that prevents us from doing so at the moment, so for now only `arm7` and `arm64` will work properly, which means no simulators will work until we fix this issue, so sorry about the delays!

@klemenzagar91 it will go away in that case yes, but if you try to archive the binary it will show the same error again because it will build for all archs when releasing. this should be OK if you just remove i386 and x86_64 from the `Valid Architectures` and the `Architectures` in your Build settings tab!

Will keep you updated as we progress on this!

mostafaberg, 2016-06-08:
This issue is resolved in our new CocoaPods library found [here](https://cocoapods.org/pods/iOSDFULibrary)

closed 2016-06-08 by mostafaberg

mostafaberg, 2018-10-19:
hi @ElizaSapir , basically the issue occurs when you build using only the active architecture, which produces a fat tile that misses the other archs. the easiest way to fix it is to avoid having to configure it.

that's why we have a new repository that uses cocoa pods to do just that.

simply create a pod file in the root of your project, and add the dependency `iOSDFULibrary` and then run the `pod install` or `pod update` command, and the project will be ready to go with no extra work on your side.

for more, info, check the new repository [here](http://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/)
and the pods page [here](https://cocoapods.org/pods/iOSDFULibrary)

if you encounter any issues, please open up a new ticket in the new repository and we'll gladly help you out.