NordicSemiconductor/IOS-DFU-Library #5: The dfu_version issue, please help.

eirstone, 2016-03-20:
I've been using the IOS-DFU-Library to update firmware, however I encounter the following issue, 
1. I got a warning below, will this affect the update I wonder, and what causes this?
WARNING: The class 'Manifest' is not key value coding-compliant for the key 'dfu_version'
 There is no support for optional type, array of optionals or enum properties.
As a workaround you can implement the function 'setValue forUndefinedKey' for this. See the unit tests for more information

2. finally I fail to update the firmware due to the reason below, the console output seems to complain about the DFU version not right. (It's expecting for 2 bytes)

Reading DFU Version number...
2016-03-20 21:16:42.021 Brushes[555:143086] 0: peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
2016-03-20 21:16:42.023 Brushes[555:143086] 5: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):05b103
2016-03-20 21:16:42.023 Brushes[555:143086] 15: Invalid value: 2 bytes expected
2016-03-20 21:16:42.024 Brushes[555:143086] 1: Disconnecting...
2016-03-20 21:16:42.024 Brushes[555:143086] 0: centralManager.cancelPeripheralConnection(peripheral)
2016-03-20 21:16:42.025 Brushes[555:143088] 0: [Callback] Central Manager did disconnect peripheral without error
2016-03-20 21:16:42.026 Brushes[555:143088] 5: Disconnected
2016-03-20 21:16:42.026 Brushes[555:142909] Error 303: Unsupported DFU Version value: 0x05b103)

Lei

philips77, 2016-03-20:
Hi, the library is compatible with the DFU bootloader from the sdk. If the DFU differs in your case you'd have to modify the library yourself. The DFU Version char from sdk returns 2 bytes, a minor and major numbers. Value 0x0100 is reserved for an app that supports buttonless jump to the bootloader, other values are described in the docs.
As to the first issue - that's true. The manifest file may have some other fields, but I assumed they will be ignored... I'll check it when I get back from vacations.

eirstone, 2016-03-20:
Hi Philips,

As you said, "If the DFU differs in your case you'd have to modify the library yourself. " I do not understand could you explain it further please?

How to let DFU version char returns 2 byte, in my case it is 0x05b103 (I do not even know where it comes from, how to specify the DFU version value?)

Lei

philips77, 2016-03-20:
What DFU implementation do you use? Is it mbed or some other 3rd party? I'm not familiar with any of those.
If your device returns this value from DFU Version char, you should know what does it matter. Then you should modify the code of the DFUVersionCharacteristic class in this library to match your requirements and build the app again.

eirstone, 2016-03-23:
Hi philips,

I manage to get firmware update worked :-)
I've created the  SDK that integrates the DFU iOS library to do firmware update. The SDK itself is able to connect peripherals and their services, and it works fine without firmware update involved. But SDK can connect to peripherals but could not find any services after update completed.

I have to reboot my app, then everything goes fine, I wonder is this normal while working with DFULibrary?

Lei

philips77, 2016-03-30:
Hi, the iOS caches the device services if it does not have the Service Changed characteristic (if not bonded) or until it receives the Service Changed indication (if bonded). If your device does not have SC and is not bonded, the iOS assumes that services will never change and may cache the services easily. To remove them you need to turn the Bluetooth OFF and ON again, or reset the iDevice.

mostafaberg, 2016-06-08:
Hi @eirstone , seems that the main issue you opened has been resolved and now you are experiencing another problem, i'll close this one for now and if you're still experiencing the latest problem please post it as a new issue and we will take it from there :)

closed 2016-06-08 by mostafaberg