NordicSemiconductor/IOS-DFU-Library #6: Library crash when update firmware, the peripheralSelector is nil

eirstone, 2016-03-29:
`func centralManager(central: CBCentralManager, didDiscoverPeripheral peripheral: CBPeripheral, advertisementData: [String : AnyObject], RSSI: NSNumber) {
        // Is this a device we are looking for?
    
        if peripheralSelector!.select(peripheral, advertisementData: advertisementData, RSSI: RSSI) {
            // Hurray!
            centralManager.stopScan()
            
            if let name = peripheral.name {
                logger.i("DFU Bootloader found with name \(name)")
            } else {
                logger.i("DFU Bootloader found")
            }
            
            self.peripheral = peripheral
            connect()
        }
    }
`
I got crash error, that is peripheralSelector is nil
unexpectedly found nil while unwrapping an Optional value

philips77, 2016-03-30:
Hi,
How did you manage to get nil there? Aren't you setting it to nil in DFUServiceInitiator somehow?
It is taken from the DFUServiceInitiator and there it can't be nil (not optional value).
The only way I see it could be when you were scanning on your own using the CentralManager while DFU is in progress. The DFUPeripheral sets manager.delegate to itself and assumes that the manager will scan only when it's ready for it (after sending SoftDevice).

The scanning in DFUPeripheral is started in ```switchToNewPeripheralAndConnect(selector:DFUPeripheralSelector)``` and stopped immediately after a device has been found. It you start scaring while DFU it will fail indeed...

eirstone, 2016-03-30:
I think I should call 'stopScan' before initializing the DFUServiceInitiator, otherwise I got nil on peripheralSelector.
`[bluetoothManager stopScan]; `

philips77, 2016-03-30:
True. I'll update the lib tomorrow to prevent this crash from happening.

philips77, 2016-03-31:
I just pushed a bug fix: https://github.com/NordicSemiconductor/IOS-DFU-Library/commit/f7de5f26182a5ef24e9a07b7a914da49ae16701d

closed 2016-03-31 by philips77