NordicSemiconductor/IOS-DFU-Library #7: Optional SoftDevice upgrade

Reggian, 2016-03-31:
Consider we have the following scenario:

        SD rev.  App rev.
    A:     1        1
    B:     2        2
    C:     2        3

Firmware A contains the SoftDevice rev 1 and the App rev 1. Firmware B contains both revisions 2. In Firmware C we want to upgrade the App to rev 3 only as it works with SoftDevice 2.
Because we would like to support A to C upgrades the C.zip must contain SD along with the App (lets call this Full Upgrade). From B to C only the App should be upgraded (called App Upgrade).

Is there a way that the library handles the updating From A to C (Full Upgrade) and B to C (App Upgrade) with a single C.zip or do we need a separate C_full.zip and C_app.zip.

I hope I presented a clear picture.

philips77, 2016-03-31:
Hi, I understand clearly. The problem is that, from what I can imagine, checking the SD version may be a security issue and thus this information is not exposed. 
But the bootloader validates the new update if it's compatible with the existing SD, which could be used to solve your problem. You'd have to try the B2C update first (app only), and if the validation fails, do A2C (SD+App). Unfortunately, in the 1.0.1 version of this library, you can't specify which content from the ZIP should be used (all or only the app), so 2 ZIP files are needed. I imagine something like "type" would be required also in the init(urlToZipFile:) of the DFUFirmware.

Reggian, 2016-04-07:
Hi,
> You'd have to try the B2C update first (app only), and if the validation fails, do A2C (SD+App).

I have tried B2C update using the BIN and DAT from the zip file. It fails with the error `Response (Op Code = 2, Status = 6) received => Operation failed`. OK. Right after that I try A2C and get `Response (Op Code = 1, Status = 2) received => Device is in invalid state`.

Any ideas?

philips77, 2016-04-07:
How about adding 1.5sec delay in between?
Maybe related to #8 ?

Reggian, 2016-04-07:
I have tried this right now, and it does not work. Unfortunately.

Reggian, 2016-04-11:
> Unfortunately, in the 1.0.1 version of this library, you can't specify which content from the ZIP should be used (all or only the app), so 2 ZIP files are needed. I imagine something like "type" would be required also in the init(urlToZipFile:) of the DFUFirmware.

I have created a [pull request](https://github.com/NordicSemiconductor/IOS-DFU-Library/pull/12) that solves this.

philips77, 2016-04-12:
Thanks, as I wrote in PR, I'll come back to it in couple of days.

philips77, 2016-04-18:
We have merged the PR. Thanks again!
It will be included in the cocoa pods release we make now.

closed 2016-04-18 by philips77