NordicSemiconductor/IOS-DFU-Library #9: [DFU][IOS 9][DOES'T GO BEYOND state = StateEnablingDfuMode]

schech-a, 2016-04-06:
Hi everyone,

I am trying to perform a DFU of my device with IOS. The process of DFU with zip file works with Nordic's BLE app, and the IOS library says that the firmware is valid.

Enabling DFU Bootloader is apparently the last successful step. After that, it goes back to the first step, and loop.

I can see with BLE Scanner that the device goes to "DfuTarg", but then goes back to the original device name.

The only information that the log gives me is : 
- Enabling DFU Bootloader...
- 0: [Callback] Central Manager did disconnect peripheral
- 5: Disconnected by the remote device
- Connecting to "MyDevice"...

I guess that it has nothing to do with the fact that the logs says that it is connecting to MyDevice instead of "DfuTarg", but it it kind of my only lead right now.

Before trying to poke around with the IOS_DFU_Library source code, does anyone face the same issue?
I am doing something wrong?

Thanks in advance :)

philips77, 2016-04-06:
Hi,
The log you sent seems correct. Could you send a bit more of it?
The iOS caches the device name, that's why it says it still MyDevice.
What version of the DFU are you using? From which SDK? Do you have the DFU Version in the app mode and in the bootloader mode? Are you sure they return value 0x0100 in app mode and 0x0(5 or more)00 in bootloader mode? That's how it checks it you are connected to the app with buttonless update afeature, or to something that can perform the actual DFU (value different than 0x0100).
If you don't have the DFU Version characteristic the library checks number of characteristics - if there is only 1 (not counting Generic Access and Generic Attribute) - it assumes it's DFU, if more - it assumes it's an app that can be switched to the DFU mode. In the latter case you may call ```initiator.forceDfu = true``` to force the DFU in such case.

schech-a, 2016-04-06:
*Hi Aleksander,First of all, thank you for your incredibly fast answer
:)Here are the answers to your questions : - DFU VERSION : 0.1- SDK : 10-
DFU MODE : With NRF Toolbox, I can see in Application Mode the DFU version,
and the buttonless update mode on the log.- App Mode : 0x0100- Bootloader
Mode : 0x0800- I have the DFU version characteristic (and tried with both
forcemode on and off)Here is a log : 2016-04-06 17:56:48.557 Tracker
Serie[5110:1801494] central DID CONNECT
49F5866A-6CC0-2A85-6CA4-0E2BBC5C93F9*

*2016-04-06 17:56:48.705 Tracker Serie[5110:1801494] *

*WARNING: The class 'Manifest' is not key value coding-compliant for the
key 'dfu_version'*

* There is no support for optional type, array of optionals or enum
properties.*

*As a workaround you can implement the function 'setValue forUndefinedKey'
for this. See the unit tests for more information*

*2016-04-06 17:56:48.711 Tracker Serie[5110:1801494] FIRMWARE NAME :
FW_TR_160404.zip*

*2016-04-06 17:56:48.712 Tracker Serie[5110:1801494] FIRMWARE PARTS : 1*

*2016-04-06 17:56:48.712 Tracker Serie[5110:1801494] FIRMWARE IS VALID*

*2016-04-06 17:56:48.714 Tracker Serie[5110:1801494] **1: Connecting to
ZTTrack...*

*2016-04-06 17:56:48.714 Tracker Serie[5110:1801494]  **0:
centralManager.connectPeripheral(peripheral, options:nil)*

*2016-04-06 17:56:48.717 Tracker Serie[5110:1801494] Connecting...*

*2016-04-06 17:56:48.717 Tracker Serie[5110:1801494] status DFU : 0*

*2016-04-06 17:56:48.717 Tracker Serie[5110:1801494] **0: [Callback]
Central Manager did connect peripheral*

*2016-04-06 17:56:48.718 Tracker Serie[5110:1801494]  **5: Connected to
ZTTrack*

*2016-04-06 17:56:48.718 Tracker Serie[5110:1801494] **1: Discovering
services...*

*2016-04-06 17:56:48.718 Tracker Serie[5110:1801494] **0:
periphera.discoverServices(nil)*

*2016-04-06 17:56:48.972 Tracker Serie[5110:1801494] **5: Services
discovered*

*2016-04-06 17:56:48.973 Tracker Serie[5110:1801494] **1: DFU Service found*

*2016-04-06 17:56:48.973 Tracker Serie[5110:1801494] **1: Discovering
characteristics in DFU Service...*

*2016-04-06 17:56:48.973 Tracker Serie[5110:1801494] **0:
peripheral.discoverCharacteristics(nil, forService:DFUService)*

*2016-04-06 17:56:49.213 Tracker Serie[5110:1801494] **5: DFU
characteristics discovered*

*2016-04-06 17:56:49.217 Tracker Serie[5110:1801494] **1: Reading DFU
Version number...*

*2016-04-06 17:56:49.217 Tracker Serie[5110:1801494] **0:
peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)*

*2016-04-06 17:56:49.273 Tracker Serie[5110:1801494] **5: Read Response
received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0100*

*2016-04-06 17:56:49.273 Tracker Serie[5110:1801494] **10: Version number
read: 0.1*

*2016-04-06 17:56:49.274 Tracker Serie[5110:1801494] **1: Enabling
notifiactions for 00001531-1212-EFDE-1523-785FEABCD123...*

*2016-04-06 17:56:49.274 Tracker Serie[5110:1801494] **0:
peripheral.setNotifyValue(true, forCharacteristic:
00001531-1212-EFDE-1523-785FEABCD123)*

*2016-04-06 17:56:49.274 Tracker Serie[5110:1801494] Starting DFU...*

*2016-04-06 17:56:49.275 Tracker Serie[5110:1801494] status DFU : 1*

*2016-04-06 17:56:49.392 Tracker Serie[5110:1801494] **1: Notifications
enabled for 00001534-1212-EFDE-1523-785FEABCD123*

*2016-04-06 17:56:49.392 Tracker Serie[5110:1801494] 1**0: DFU Control
Point notifications enabled*

*2016-04-06 17:56:49.393 Tracker Serie[5110:1801494] **15: Application with
buttonless update found*

*2016-04-06 17:56:49.394 Tracker Serie[5110:1801494] **1: Writing to
characteristic 00001531-1212-EFDE-1523-785FEABCD123...*

*2016-04-06 17:56:49.394 Tracker Serie[5110:1801494] **0:
peripheral.writeValue(0x0104, forCharacteristic:
00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)*

*2016-04-06 17:56:49.395 Tracker Serie[5110:1801494] Enabling DFU
Bootloader...*

*2016-04-06 17:56:49.395 Tracker Serie[5110:1801494] status DFU : 2*

*2016-04-06 17:56:50.161 Tracker Serie[5110:1801494] **0: [Callback]
Central Manager did disconnect peripheral*

*2016-04-06 17:56:50.162 Tracker Serie[5110:1801494] **5: Disconnected by
the remote device*

*2016-04-06 17:56:50.162 Tracker Serie[5110:1801494] **1: Connecting to
ZTTrack...*

*2016-04-06 17:56:50.162 Tracker Serie[5110:1801494] **0:
centralManager.connectPeripheral(peripheral, options:nil)*

*2016-04-06 17:57:22.073 Tracker Serie[5110:1801494] 0**: [Callback]
Central Manager did connect peripheral*

*2016-04-06 17:57:22.073 Tracker Serie[5110:1801494] **5: Connected to
ZTTrack*

*2016-04-06 17:57:22.074 Tracker Serie[5110:1801494] **1: Discovering
services...*

*2016-04-06 17:57:22.074 Tracker Serie[5110:1801494] **0:
periphera.discoverServices(nil)*

*2016-04-06 17:57:22.505 Tracker Serie[5110:1801494] **5: Services
discovered*

*2016-04-06 17:57:22.505 Tracker Serie[5110:1801494] **1: DFU Service found*

*2016-04-06 17:57:22.506 Tracker Serie[5110:1801494] **1: Discovering
characteristics in DFU Service...*

*2016-04-06 17:57:22.506 Tracker Serie[5110:1801494] **0:
peripheral.discoverCharacteristics(nil, forService:DFUService)*

*2016-04-06 17:57:22.743 Tracker Serie[5110:1801494] **5: DFU
characteristics discovered*

*2016-04-06 17:57:22.744 Tracker Serie[5110:1801494] **1: Reading DFU
Version number...*

.....

IT LOOPS ON AND ON....

.....

*2016-04-06 17:57:22.925 Tracker Serie[5110:1801494] Enabling DFU
Bootloader...*

*2016-04-06 17:57:22.926 Tracker Serie[5110:1801494] status DFU : 2*

*2016-04-06 17:57:23.690 Tracker Serie[5110:1801494] **0: [Callback]
Central Manager did disconnect peripheral*

*2016-04-06 17:57:23.690 Tracker Serie[5110:1801494] *

*5: Disconnected by the remote deviceThanks for you help Aleksander!Alexis*

schech-a, 2016-04-06:
Ok I found out what is going on...

As I was not sure how the lib was getting the new DFU TARG UUID because I was only giving you the APP UUID. I tried to put manually in DFU TARG, and give you the CBPeriphal of the DFUTARG and it worked... As simple as that.

Thanks

closed 2016-04-06 by schech-a