NordicSemiconductor/IOS-DFU-Library #10: Framework not existing

sanderdb01, 2016-04-07:
I am trying to install the DFU framework into my XCode project and I am have a lot of difficulty. For the first option on the install page it say to just copy the DFULibrary.framework file to the project, but the file does not exist in the repository. The second option  says to include the library as source code, but when I try adding it to the project, the framework file is always red and says it does not exist. I then tried creating my own framework file based on the library code, and that did not work either. I was wondering if someone could help me add this to my code. Thanks.

link to the steps I tried: https://github.com/NordicSemiconductor/IOS-DFU-Library/tree/master/documentation

philips77, 2016-04-07:
Hi, you should be able to select Project->Select scheme (?) and then DFU library, then Project->Build. This will build the framework and you'll find it in Products in the lib project.
Tomorrow I'll push some configuration changes so that it will work added as a framework only, not embedded binaries. But you should be able to follow these steps from the link.

philips77, 2016-04-08:
One more thing, the ZIP with framework was released in releases, not in code: https://github.com/NordicSemiconductor/IOS-DFU-Library/releases.
But it was not build using Archive, but just debug build. Like I said I'll do some changes today.

sanderdb01, 2016-04-08:
Thanks for the follow up, I really appreciate it. I have been able to add the framework to my project now, but I have now ran into another issue. So, in my simulator I can build and run my project. But when I try and run the project on my device, all of the protocols that I have for the DFU say "DFUProgressDelegate is unavailable: cannot find Swift declaration for this protocol", and it says that for all the the protocols. I have been trying to search the interwebs for a solution, but to no avail. Any ideas? Thanks a bunch.

sanderdb01, 2016-04-08:
OK, I have s solution to my comment above. Basically I had to remove the framework from my project totally (in the trash), clean the project, build the framework again making sure it is for a device, and then add it back to my project and it will work. It is an old XCode bug they apparently still have not worked out yet.

philips77, 2016-04-08:
Very good!

closed 2016-04-08 by philips77