NordicSemiconductor/IOS-DFU-Library #11: Issue with init packet and device disconnecting when starting DFU

sanderdb01, 2016-04-08:
So, I am using my iOS program to do DFU. I finally have it to a point where I am getting feedback from the device. The way my nRF application is set up, I have to manually restart the nordic chip into the boot loader by sending the DFU characteristic a value of 0x0104, which it successfull reboots into the DFU. I then reconnect to the module while I am in DFU mode and try to start the process, but I get the following logger output (some of this is my own print statements, so please disregard):

Scanning for DFU peripheral
Saved DFU peripheral detected, connecting to it
Logger Message: Connecting to DfuTarg...
Logger Message: centralManager.connectPeripheral(peripheral, options:nil)
State of the DFU Upload: 0
Logger Message: [Callback] Central Manager did connect peripheral
Logger Message: Connected to DfuTarg
Logger Message: Discovering services...
Logger Message: periphera.discoverServices(nil)
Logger Message: Services discovered
Logger Message: DFU Service found
Logger Message: Discovering characteristics in DFU Service...
Logger Message: peripheral.discoverCharacteristics(nil, forService:DFUService)
Logger Message: DFU characteristics discovered
Logger Message: Reading DFU Version number...
Logger Message: peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
Logger Message: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0800
Logger Message: Version number read: 0.8
Logger Message: Disconnecting...
Logger Message: centralManager.cancelPeripheralConnection(peripheral)
Logger Message: [Callback] Central Manager did disconnect peripheral without error
Logger Message: Disconnected
DFU Error: The init packet is required by the target device  

So, any ideas why I am automatically disconnecting from the device, and why does it say that the init packet is required.

Also, I saved my .hex file to a website (url), and that is where I am getting the firmware from. It is not password protected, but a straight link to the .hex file. Thanks a bunch

philips77, 2016-04-08:
Hi, 
This is easy, so you get reply right away.
Since SDK 7.1 (as far as I remember) the DFU requires the init packet to be sent together with the firmware file. The service disconnect after checking the DFU version (0.8, or 0x0800) which means that this init file is required. To send it you may either set it using DFUFirmware.init(urlToHexOrBinFile:urlToDatFile:type), or use the nrf util to create a Distribution Packet, which is a Zip file containing bin, dat and manifest files). Zip may also contain a SoftDevice and/or Bootloader. Please read https://github.com/NordicSemiconductor/nRF-Master-Control-Panel/blob/master/init%20packet%20handling/README.md or DFU documentation for more info.

Hope that helped,
Good night :)

philips77, 2016-04-08:
The links on Github link I gave may be outdated. Please see current DFU docs on infocenter.
The nrf util is available on its own github account (search in NordicSemiconductor) or included in PC version of Master Control Panel since 3.7.+, I guess...

sanderdb01, 2016-04-08:
Ok cool, thanks for being so quick about it =). Alright, so I talked to the other engineer and it turns out (like you said) I need more than just the hex file. So we created a make file with our nRF program and created a zip file with everything I need and put it on the website. However, now I am getting this error:

Error while creating ZIP stream: The operation couldn’t be completed. (DFULibrary.ZipError error 0.)

Any ideas on this one? 

Again, thank you so much for the help. This needs to be done end of business and im struggling.

philips77, 2016-04-08:
Check if you have all required data in the manifest file: app version, dev revision, dev type, required SD, dat and bin file. And not the DFU version, as the Parsing may fail.

Place the manifest file here if doesn't work.

philips77, 2016-04-15:
Hi, did you manage to fix your issue?

sanderdb01, 2016-04-15:
Yep, got it all figured out. Thanks a bunch =)

closed 2016-04-15 by philips77

kimyd, 2016-09-19:
Hello, I got same issue. I can update by using nrf tool box (DFU) with generated zip file. But I got same error message by using URL (http:///myfw.zip) with this lib. 
Do I have to delete dfu_version in the manifest file? 
Thank you.