NordicSemiconductor/IOS-DFU-Library #14: Wrong archiving app that includes DFU Library.

klemenzagar91, 2016-04-16:
[image]
As you see, if I use the latest code from your repository, the app is not archived correctly. It is shown in the Other items section and the button Upload to app store is disabled.

If I go into the build settings for DFU Library, and set Skip Install to YES, then the archive is shown in the iOS app section. And in that case the Upload app store button is enabled.

But in that case there is another problem. If I try to upload the app to itunes connect the procces gets completed, but I get two warnings on my email. The build is somehow corrupted and can be used inside the itunes connect. Here is the content from these two emails:

----------------------
Dear developer,

We have discovered one or more issues with your recent delivery for "Scoutee Baseball Radar". To process your delivery, the following issues must be corrected:

Invalid Bundle - This app includes files that were built with bitcode enabled but without doing an Archive or Install build, possibly from a library or framework that was imported but not built as part of the project. Full bitcode content is only generated in an Archive or Install build.

Once these issues have been corrected, you can then redeliver the corrected binary.

Regards,

The App Store team

--------------------------------

Dear developer,

We have discovered one or more issues with your recent delivery for "Scoutee Baseball Radar". Your delivery was successful, but you may wish to correct the following issues in your next delivery:

Too many symbol files - These symbols have no corresponding slice in any binary [CA478669-6722-3C00-A065-DAFE23F05630.symbols, E549BE4F-6535-3F18-BF84-7C6B492B902F.symbols]

After you’ve corrected the issues, you can use Xcode or Application Loader to upload a new binary to iTunes Connect.

Regards,

The App Store team
------------------------

I event try adding DFU library to clean project and uploading it to itunes connect, but there was no luck either. Whats your advice. Do this library works for you. Did you try to download it from github and use it in the clean project and try to upload it on itunes connect?

mostafaberg, 2016-04-16:
Hi @klemenzagar91 , sorry to see those issues you're having, I'll take a look into this on Monday!,

How do you have your project setup? Are you adding the DFU lib to a project's workspace or are you building the framework and copying the built framework file to your project ?

klemenzagar91, 2016-04-16:
Hi, thanks for taking care of this. Im using library as part of the workspace.

mostafaberg, 2016-04-17:
I see, when we did the submission we actually built the library separately then copied the built framework into the project, but remember not to just build but to actually archive th library, then open the organizer and then export it from there

I don't have a Mac at the moment so I can't be very helpful I'm sorry.

philips77, 2016-04-17:
We could update the Readme on Github regarding this.

mostafaberg, 2016-04-17:
That's true, I'll get that done!

klemenzagar91, 2016-04-17:
Hey, it works when I archives your library and then add the framework to my project. This is really fantastic.
thumbs up

The only issue that remains is that now the app cant be run on simulator. Hope you'll fix this in the future. But this is not vital. If it helps, I can tell you that running on simulator works if I imported your library as a project in a workspace.

mostafaberg, 2016-04-17:
@klemenzagar91 That's perfect !, We will hopefully get it running on simulator as soon as possible, on the long run it will probably be available as a cocoapod too to make builds even easier.

Thanks for the tip about the workspace/simulator, will check that out for sure, by the way, which version of XCode are you using ?

klemenzagar91, 2016-04-17:
Oh, cocoapods will be great. It'll make the integration much simplified. I'm using latest xcode version 7.3.

mostafaberg, 2016-04-17:
Ok, I'm using 7.3 too, that's perfect !, you'll surely be updated when we fix it soon, sorry about the delays:)

philips77, 2016-04-17:
Delays? You reply on Sunday evening ;)

mostafaberg, 2016-04-17:
@philips77 That's true, still too slow, should've responded before he even opened the issue :smile:

klemenzagar91, 2016-04-17:
Your availability is great, no doubt.
I can send you errors that I get when I try to run it on simulator. Just tell me if you need it.

mostafaberg, 2016-04-20:
We have a cocoa pod released now that you can find [here](https://cocoapods.org/pods/iosdfulibrary) hope this helps, will update the master control panel app to include the changes soon and package it for release, let us know if you have any more issues, for now will close this issue 👍 thanks for the help!

closed 2016-06-08 by mostafaberg