NordicSemiconductor/IOS-DFU-Library #16: The library does not handle Central Manager states

Reggian, 2016-04-20:
If Bluetooth is **reset/turned off** during the upgrade process I think either of the two should happen:
a) **fail** with error "Bluetooth was turned off..."
b) **auto resume/restart** after it is on again

I think that option a) is more likely to get implemented and should get implemented soon because the DFU process now "freezes" without a state change...

What is your word on this?

Best,
R.

mostafaberg, 2016-04-20:
Seems like a bug or we are not handling interruptions properly, auto resume will not be possible since it depends on the DFU device's state, so the best is that we can fail with an error, will look into that!
thanks

hudanavroz, 2016-07-08:
It is searching again for connection after completing one part of update.It is not completing upload process.

halijoe, 2016-08-10:
So how to solve this  bug? @mostafaberg

halijoe, 2016-08-10:
My problem is when an upgrade process completed,my app can't scan any peripherals,and the state of the centralManager object is still poweredOn.How to let it continue to scan new periperals? @mostafaberg

mostafaberg, 2016-08-10:
This seems to be another issue not related to this one can you open up a new issue so we can take it there ? also put there any log output at the end of the DFU process and any extra information you have, thanks

philips77, 2016-08-10:
Hi, did you restore your delegate in central manager? The lib overrides existing one with its own.

halijoe, 2016-08-11:
That it!Thanks!

halijoe, 2016-08-11:
That's it!Thanks!

morgz, 2016-09-08:
The library 'hijacks' the central manager you give it. I am going about creating 2 central managers. I use one for the DFU, and the other for monitoring bluetooth events. Managing 2 is a bit easier using RxSwift and this bluetooth library: https://github.com/Polidea/RxBluetoothKit

mostafaberg, 2016-09-08:
@morgz I think it's better to just wait until the library updates the state to `completed`, then you can take back the delegate there and resume as you should.

The main reason we are taking the delegate away is that you shouldn't be able to listen to it since this is a DFU implementation and you do not know the entire protocol, sometimes we will disconnect the device intentionally for many reasons, and if you get that callback  you might implement a `reconnect` feature which will eventually break the process.

We do however return proper status updates to the delegate so you'll be able to monitor the progress and any errors. once the state has been returned to complete, our plan is to automatically return the delegate back at this point.

you can do that manually until we implement it.

mostafaberg, 2016-09-08:
@morgz I also don't know if having two Central managers will work or not

morgz, 2016-09-08:
Thanks for the response. According to Apple Docs you can have more than one. 
My reason for wanting to monitor the state of the bluetooth delegate is related to this issue. If I turn off Bluetooth mid update, then everything just hangs. I would have expected to get an error through to the DFUServiceDelegate.

mostafaberg, 2016-09-08:
@morgz You should receive an error, this has to be a bug, can you open up a new ticket for that bug and we take it there ?

I didn't mean that you can't have more than one central manager, you can do that but I don't know how that will work if it will

morgz, 2016-09-08:
OK Will do. It's obviously an independent case, but the original description of this issue explains my problem.

mostafaberg, 2016-09-08:
True, the solution will work for you but it's the wrong solution, just covering up a bug, will take a look into that issue if you open up a ticket sometime so i have a reminder to fix it.

morgz, 2016-09-08:
#39

mostafaberg, 2016-10-05:
[This issue was resloved](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library) in our new cocoapods library, I recommend switching to this library instead since we will be deprecating this standalone library

closed 2016-10-05 by mostafaberg