NordicSemiconductor/IOS-DFU-Library #17: Both centralManager and peripheral's delegate set to be nil when DFU finished

allencheng17, 2016-04-22:
Both centralManager and peripheral's delegate set to be nil when DFU finished which leads my original centralManager and peripheral's delegate to be nil. Then connect timeout as centralManager' delegate is nil, discoverService timeout too as peripheral's delegate is nil too.
Currently my workaround is to set the delegates back manually once DFU finished.
So is that possible that the library could set the centralManager and peripheral's delegate back when DFU finished?

philips77, 2016-04-22:
I don't know if we should do it. There are the didStateChangedTo(Completed) and didErrorOccur(...) delegates which can be used for that purpose, and perhaps you're doing it. The delegates can be long released after the DFU is done.. But on other hand, if we use weak references it could work. How do you think @mostafaberg?

mostafaberg, 2016-04-25:
@philips77 I will have to test that tomorrow and confirm why this happens, ARC should handle this kind of issue so it shouldn't really clean out those vars, depending on how the code is handling the variables, a simple weak reference might fix it, but it all depends on the implementation, will check and verify with you tomorrow :+1:

mostafaberg, 2016-04-26:
I agree with @philips77 we can't do that right now, this seems to be by design, imagine that your device is in DFU and being flashed, you shouldn't be getting any delegate calls at that time, even if you did, any accidental code that will try to communicate with the device will probably break the flow, for the time being I recommend you do the following as @philips77 recommended:

 - Hold references to the delegates before you start DFU
 - When `didStateChangedTo(Completed)` or `didErrorOccur(...)` get called, set those delegates back.

On the long run we can probably do those steps within the library, but it would be a bad idea to have your delegate interfere during DFU, so you will not get callbacks for that even after we "fix" this issue, we will only set the delegates back again upon completion.

If you tell us what you are trying to achieve by having the delegates maybe we can give you some tips on another way to achieve it, let us know!

mostafaberg, 2016-06-08:
Hi @allencheng17 just checking in to see if you've managed to get your issue resolved ? we might address this inconvenience in a later version and will keep you updated if we decide to do so, it'll be given a higher priority if we see it as a serious bug

closed 2016-06-08 by mostafaberg