NordicSemiconductor/IOS-DFU-Library #19: Dfu Transfer error about the hex file after DFU_LIB_3.0

kc4712, 2016-04-26:
i development work on firmware update ios(developemen language:SWIFT) to nrf51822, DFU3.0 code previously had a works fine
but, now i use to DFU Library 3.0 LIB, i don't understand about this condition.
how i should about this conditions?

i report that condition 

//////// ------------before fix the lib3.0 code
//XCODE LOG
part:1/totalParts:1, progress:87%, <avgSpeedBPS:2492.32686410435/currentSpeedBPS:5188.08089554085>
packetsToSendNow2 : bytesLeft9000 : packetLength20
bytesSent65740 > progress87 :  bytesTotal74720

part:1/totalParts:1, progress:87%, <avgSpeedBPS:2492.7248767743/currentSpeedBPS:5245.17476395923>
packetsToSendNow1 : bytesLeft8980 : packetLength20
bytesSent65760 > progress87 :  bytesTotal74720

part:1/totalParts:1, progress:88%, <avgSpeedBPS:2493.05340101518/currentSpeedBPS:4398.50457488923>
this sendline <DFULibrary.DFUService: 0x13c67c290>
bytesSent: 65760 // bytesReceived: 224 // firmware.size:74720
LogLevel: Disconnecting...
LogLevel: centralManager.cancelPeripheralConnection(peripheral)
this error line CBCentralManagerState
LogLevel: [Callback] Central Manager did disconnect peripheral without error
LogLevel: Disconnected
Error 308: 65760 bytes were sent while 224 bytes were reported as received

///// ---------------  fix after DFUService.swift at this line
//CODE  DFUService.swift
onPacketReceiptNofitication: {
  bytesReceived in
  // Each time a PRN is received, send next bunch of packets
  if !self.paused && !self.aborted {
     print("this sendline \(self)")
     print("bytesSent: \(self.dfuPacketCharacteristic!.bytesSent) 
     // bytesReceived: \(bytesReceived) 
     // firmware.size:\(firmware.data.length)")
     let bytesSent = self.dfuPacketCharacteristic!.bytesSent
     //if bytesSent == bytesReceived {
          self.dfuPacketCharacteristic!.sendNext(number, packetsOf: firmware, andReportProgressTo: progressDelegate)
     /*} else {
           // Target device deported invalid number of bytes received
           report(error:DFUError.BytesLost, withMessage: "\(bytesSent) bytes were sent while \(bytesReceived) bytes were reported as received")
     }*/
     } else if self.aborted {
           // Upload has been aborted. Reset the target device. It will disconnect automatically
                                self.sendReset(onError: report)
     }

////------------ fix after the lib3.0 have a works fine
//XCODE LOG

part:1/totalParts:1, progress:87%, <avgSpeedBytesPerSecond:2524.85029737374/currentSpeedBytesPerSecond:11068.2253595461>
packetsToSendNow2 : bytesLeft9000 : packetLength20
bytesSent65740 > progress87 :  bytesTotal74720
part:1/totalParts:1, progress:87%, <avgSpeedBytesPerSecond:2525.44276072691/currentSpeedBytesPerSecond:11031.4731893349>
packetsToSendNow1 : bytesLeft8980 : packetLength20
bytesSent65760 > progress87 :  bytesTotal74720
part:1/totalParts:1, progress:88%, <avgSpeedBytesPerSecond:2526.0224293044/currentSpeedBytesPerSecond:10288.0367928867>
this sendline <DFULibrary.DFUService: 0x14f66ff60>
bytesSent: 65760 // bytesReceived: 224 // firmware.size:74720
packetsToSendNow12 : bytesLeft8960 : packetLength20
bytesSent65780 > progress88 :  bytesTotal74720
part:1/totalParts:1, progress:88%, <avgSpeedBytesPerSecond:2520.29955832441/currentSpeedBytesPerSecond:298.289192956504>
packetsToSendNow11 : bytesLeft8940 : packetLength20
bytesSent65800 > progress88 :  bytesTotal74720
part:1/totalParts:1, progress:88%, <avgSpeedBytesPerSecond:2520.87247671513/currentSpeedBytesPerSecond:9990.00595450756>
packetsToSendNow10 : bytesLeft8920 : packetLength20
bytesSent65820 > progress88 :  bytesTotal74720
part:1/totalParts:1, progress:88%, <avgSpeedBytesPerSecond:2521.47225759081/currentSpeedBytesPerSecond:11607.718545681>
packetsToSendNow9 : bytesLeft8900 : packetLength20
bytesSent65840 > progress88 :  bytesTotal74720
part:1/totalParts:1, progress:88%, <avgSpeedBytesPerSecond:2522.05978600003/currentSpeedBytesPerSecond:10816.6828922343>
packetsToSendNow8 : bytesLeft8880 : packetLength20
bytesSent65860 > progress88 :  bytesTotal74720

packetsToSendNow2 : bytesLeft40 : packetLength20
bytesSent74700 > progress99 :  bytesTotal74720
part:1/totalParts:1, progress:99%, <avgSpeedBytesPerSecond:2524.96768585912/currentSpeedBytesPerSecond:11061.657545988>
packetsToSendNow1 : bytesLeft20 : packetLength20
bytesSent74720 > progress99 :  bytesTotal74720
part:1/totalParts:1, progress:100%, <avgSpeedBytesPerSecond:2525.49151239219/currentSpeedBytesPerSecond:11217.3409554374>
LogLevel: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100301
LogLevel: Response (Op Code = 3, Status = 1) received
LogLevel: Upload completed in 29.64 seconds
LogLevel: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LogLevel: peripheral.writeValue(0x<04>, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
State
Validating
LogLevel: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LogLevel: Validate Firmware (Op Code = 4) request sent
LogLevel: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100401
LogLevel: Response (Op Code = 4, Status = 1) received
LogLevel: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LogLevel: peripheral.writeValue(0x<05>, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
State
Disconnecting
LogLevel: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LogLevel: Activate and Reset (Op Code = 5) request sent
LogLevel: [Callback] Central Manager did disconnect peripheral
LogLevel: Disconnected by the remote device
State
Completed

philips77, 2016-04-26:
Looks like ```bytesReceived``` is uint16, instead of uint32. Or something wrong with decoding this value. It failed exactly at 0xFFFF. I'll have a look in a moment.

philips77, 2016-04-26:
The library code looks fine, after a quick glimpse. Both values are Int, and decoded as UInt32. I will have to test it at work. 
Are you sure you send a correct value in the Packet Receipt Notification? Either the lib has problems with decoding a value longer than 16b, or it's a bug on the DFU target side.

kc4712, 2016-04-26:
current DFU lib3.0 used a Swift, and i studying about swift and iOS....
but i hope to you look at the my DFU LIB(base on nrf tool box 2.5)log 

2016-04-26 15:24:57.825 testDfu[2289:817155] HexFileSize: 210218 and BinFileSize: 74720
2016-04-26 15:24:57.825 testDfu[2289:817155] Number of Packets 3736 Bytes in last Packet 20
2016-04-26 15:24:57.826 testDfu[2289:817155] DFUOperationsdetails enableNotification
2016-04-26 15:24:57.826 testDfu[2289:817155] DFUOperationsdetails startDFU
2016-04-26 15:24:57.826 testDfu[2289:817155] DFUOperationsdetails writeFileSize 4
2016-04-26 15:24:57.827 testDfu[2289:817155] APPLICATION firmwareSize 74720
2016-04-26 15:24:57.828 testDfu[2289:817155] performDFUOnFile
2016-04-26 15:24:57.935 testDfu[2289:817155] didWriteValueForCharacteristic 00001531-1212-EFDE-1523-785FEABCD123 and value <100306>
2016-04-26 15:24:57.994 testDfu[2289:817155] didUpdateValueForCharacteristic
2016-04-26 15:24:57.994 testDfu[2289:817155] received notification <100103>
2016-04-26 15:24:57.994 testDfu[2289:817155] processDFUResponse
2016-04-26 15:24:57.994 testDfu[2289:817155] processsRequestedCode
2016-04-26 15:24:57.994 testDfu[2289:817155] Requested code is StartDFU now processing response status
2016-04-26 15:24:57.994 testDfu[2289:817155] processStartDFUResponseStatus
2016-04-26 15:24:57.995 testDfu[2289:817155] device has old DFU. switching to old DFU...
2016-04-26 15:24:57.996 testDfu[2289:817155] stringFileExtension 진입 case Hex
2016-04-26 15:24:58.004 testDfu[2289:817155] HexFileSize: 210218 and BinFileSize: 74720
2016-04-26 15:24:58.004 testDfu[2289:817155] Number of Packets 3736 Bytes in last Packet 20
2016-04-26 15:24:58.004 testDfu[2289:817155] onFileOpened file size: 74720
2016-04-26 15:24:58.004 testDfu[2289:817155] DFUOperationsdetails enableNotification
2016-04-26 15:24:58.004 testDfu[2289:817155] DFUOperationsdetails startOldDFU
2016-04-26 15:24:58.005 testDfu[2289:817155] writeFileSizeForOldDFU file:///var/containers/Bundle/Application/9B60D241-D3A7-42B7-832E-7FDC1BD2EED7/testDfu.app/firmwares/aaa.hex
2016-04-26 15:24:58.005 testDfu[2289:817155] performOldDFUOnFile
2016-04-26 15:24:58.114 testDfu[2289:817155] didWriteValueForCharacteristic 00001531-1212-EFDE-1523-785FEABCD123 and value <100103>
2016-04-26 15:24:58.174 testDfu[2289:817155] didUpdateValueForCharacteristic
2016-04-26 15:24:58.174 testDfu[2289:817155] received notification <100101>
2016-04-26 15:24:58.174 testDfu[2289:817155] processDFUResponse
2016-04-26 15:24:58.175 testDfu[2289:817155] processsRequestedCode
2016-04-26 15:24:58.175 testDfu[2289:817155] Requested code is StartDFU now processing response status
2016-04-26 15:24:58.175 testDfu[2289:817155] processStartDFUResponseStatus
2016-04-26 15:24:58.175 testDfu[2289:817155] successfully received startDFU notification
2016-04-26 15:24:58.175 testDfu[2289:817155] DFUOperationsdetails enablePacketNotification
2016-04-26 15:24:58.176 testDfu[2289:817155] DFUOperationsdetails receiveFirmwareImage
2016-04-26 15:24:58.176 testDfu[2289:817155] DFUOperations: progress: 0%
2016-04-26 15:24:58.235 testDfu[2289:817155] didWriteValueForCharacteristic 00001531-1212-EFDE-1523-785FEABCD123 and value <100101>
2016-04-26 15:24:58.296 testDfu[2289:817155] didWriteValueForCharacteristic 00001531-1212-EFDE-1523-785FEABCD123 and value <100101>
2016-04-26 15:24:58.384 testDfu[2289:817155] didUpdateValueForCharacteristic
2016-04-26 15:24:58.385 testDfu[2289:817155] received notification <11c80000 00>
2016-04-26 15:24:58.385 testDfu[2289:817155] processDFUResponse
2016-04-26 15:24:58.385 testDfu[2289:817155] received Packet Received Notification
.
.
.
2016-04-26 15:25:28.625 testDfu[2289:817155] didUpdateValueForCharacteristic
2016-04-26 15:25:28.626 testDfu[2289:817155] received notification <11400000 00>
2016-04-26 15:25:28.626 testDfu[2289:817155] processDFUResponse
2016-04-26 15:25:28.626 testDfu[2289:817155] received Packet Received Notification
2016-04-26 15:25:28.627 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.627 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.628 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.629 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.629 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.630 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.633 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.635 testDfu[2289:817155] DFUOperations: progress: 87%
2016-04-26 15:25:28.636 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.637 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.715 testDfu[2289:817155] didUpdateValueForCharacteristic
2016-04-26 15:25:28.716 testDfu[2289:817155] received notification <11080100 00>
2016-04-26 15:25:28.716 testDfu[2289:817155] processDFUResponse
2016-04-26 15:25:28.716 testDfu[2289:817155] received Packet Received Notification
2016-04-26 15:25:28.717 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.717 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.718 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.719 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.719 testDfu[2289:817155] DFUOperations: progress: 88%
2016-04-26 15:25:28.721 testDfu[2289:817155] DFUOperations: progress: 88%
[... 19 lines cut ...]
[... cut ...]
2016-04-26 15:25:33.036 testDfu[2289:817155] Requested code is Validate Firmware now processing response status
2016-04-26 15:25:33.037 testDfu[2289:817155] processValidateFirmwareResponseStatus
2016-04-26 15:25:33.037 testDfu[2289:817155] succesfully received notification for ValidateFirmware
2016-04-26 15:25:33.037 testDfu[2289:817155] DFUOperationsdetails activateAndReset
2016-04-26 15:25:33.038 testDfu[2289:817155] upload time in sec: 35
2016-04-26 15:25:33.290 testDfu[2289:817155] didDisconnectPeripheral

kc4712, 2016-04-26:
DFU lib 3.0 transfer to DFU Target about not support DFU V.2 it switch to DFU V.1 and previously version(1.6, 2.5) same switch to performOldDFU

that work's become to this condition?

philips77, 2016-04-26:
I've tested the DFU Library and it works fine. I think the problem is on the DFU target side.
You are not trying to send a firmware longer than 65535 bytes (the size of uint16). Every Packet Receipt Notification sends number of bytes received by the DFU target.
Check at this code that you posed:

> part:1/totalParts:1, progress:88%, avgSpeedBPS:2493.05340101518/currentSpeedBPS:4398.50457488923
this sendline
**bytesSent: 65760** // **bytesReceived: 224** // **firmware.size:74720**
LogLevel: Disconnecting...
LogLevel: centralManager.cancelPeripheralConnection(peripheral)
this error line CBCentralManagerState
LogLevel: [Callback] Central Manager did disconnect peripheral without error
LogLevel: Disconnected
Error 308: 65760 bytes were sent while 224 bytes were reported as received

It looks like the byte counter has rolled over and start counting from 0.
In the library we check the reported number of bytes with number of bytes sent. I don't know if such a check was in nRF Toolbox 2.5, in the previous implementation.

I decode the number of bytes received in [here](https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/master/DFULibrary/DFUControlPoint.swift#L196). Please, put a breakpoint there and make sure you get correct bytes (remember about the Little Endian). You may also print the **data** there and post values here.

kc4712, 2016-04-27:
i try to breakpoint this line data.getBytes(&bytesReceived, range: NSRange(location: 1, length: 4))

print out this
->     Printing descrptiion of data: <11f00000 00>

what i try this line? 
data.getBytes(&bytesReceived, range: NSRange(location:1 , length:4))

kc4712, 2016-04-27:
i think bytesReceived over 65536 restarted 0, 1, 2, .......
that problem to start again, i was modified as if routine. can i use these fine?, 

///------DFUService.swift

let bytesTotal = firmware.data.length
//if bytesSent == bytesReceived {     ////   
if bytesSent <= bytesTotal {
     self.dfuPacketCharacteristic!.sendNext(number, packetsOf: firmware, andReportProgressTo: progressDelegate)
} else {
// Target device deported invalid number of bytes received
     report(error:DFUError.BytesLost, withMessage: "(bytesSent) bytes were sent while (bytesReceived)      bytes were reported as received")
}

philips77, 2016-04-27:
As you see the bug is on the firmware side. It should continue to send a correct number of bytes received, not reset to 0.
But if you don't want to fix it, or can't, you can have leave the 'workaround' in your application. But then you will have to remember, that only your app can update your firmware. The current version of Android DFU library doesn't do that check, so it will also work on Android. But I highly recommend to fix the bug on the firmware side.

closed 2016-06-08 by mostafaberg