NordicSemiconductor/IOS-DFU-Library #20: About DFU-Library increases the size of app

eirstone, 2016-04-29:
Hi, 

I have an issue regarding to the size of framework. 

I create a framework written in Objective-C that integrates the IOS-DFU-Library to update firmware.However I can not directly add IOS-DFU-Library into the framework, since it causes an issue in main app which gives out an error of "dfu library is not loaded", the dynamic framework can not embed another dynamic framework , right? So I copy all of the Swift files inside IOS-DFU-Library to the framework I am working on, and manages to work, now I can upgrade the firmware successfully. 

The question is that the main app increase the size dramatically after integrating the framework, I think this is caused by swifts files inside framework, is that possibly to reduce the size, and If so how to do it? 

Lei

mostafaberg, 2016-05-01:
This is not really an issue related to the library, and yes you shouldn't use nested frameworks, and we cannot support the library if you package it that way, I would recommend using cocoapods and add our library as a dep. This will fix the first issue, when you build your framework it will be missing the framework, you simply add both dependencies into your target app, there's really a million ways to do it and it's up to you how to package that stuff. I would ask in a more related group on how to do that, I cannot comment on the binary size increase since I don't know what's in your framework but a big increase doesn't sound right

philips77, 2016-05-01:
The increased size may be caused by the swift libraries added to your app. If you were using only Objective-C, and not you added this library, XCode will add them in order to run the code properly. But the shouldn't be more than 2-5 MB on the final file that is downloadable from AppStore. Your archive may be bigger, but I think it includes builds for different platforms.
If you want to stay with Objective-C, there is a O-C implemention of DFU in nRF Toolbox, version 2.5. Check this on GitHub. However, that release is no longer supported and all new features will be added only to this lib, in swift.

NSLogSong, 2016-05-18:
@philips77  hi,i have a error when i dfu update complete,but yesterday it performance good,the log like this:
2016-05-18 20:20:12.522 WristbandAssist[551:94993] Progress: 100% (part 1/1). Speed: 110521.844532 bps, Avg speed: 2037.557484 bps
2016-05-18 20:20:12.581 WristbandAssist[551:94993] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100301
2016-05-18 20:20:12.581 WristbandAssist[551:94993] 10: Response (Op Code = 3, Status = 1) received
2016-05-18 20:20:12.582 WristbandAssist[551:94993] 10: Upload completed in 66.63 seconds
2016-05-18 20:20:12.582 WristbandAssist[551:94993] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-05-18 20:20:12.583 WristbandAssist[551:94993] 0: peripheral.writeValue(0x04, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-05-18 20:20:12.584 WristbandAssist[551:94993] Uploading...
2016-05-18 20:20:12.641 WristbandAssist[551:94993] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-05-18 20:20:12.641 WristbandAssist[551:94993] 10: Validate Firmware (Op Code = 4) request sent
2016-05-18 20:20:12.792 WristbandAssist[551:94993] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100401
2016-05-18 20:20:12.793 WristbandAssist[551:94993] 10: Response (Op Code = 4, Status = 1) received
2016-05-18 20:20:12.793 WristbandAssist[551:94993] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-05-18 20:20:12.794 WristbandAssist[551:94993] 0: peripheral.writeValue(0x05, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-05-18 20:20:12.795 WristbandAssist[551:94993] Disconnecting...
2016-05-18 20:20:13.555 WristbandAssist[551:94993] 10: Activate and Reset (Op Code = 5) request sent
2016-05-18 20:20:13.556 WristbandAssist[551:94993] 15: Device disconnected before sending ACK
2016-05-18 20:20:13.556 WristbandAssist[551:94993] 15: Error 14: Unlikely error.
2016-05-18 20:20:13.557 WristbandAssist[551:94993] 0: [Callback] Central Manager did disconnect peripheral
2016-05-18 20:20:13.557 WristbandAssist[551:94993] 5: Disconnected by the remote device
2016-05-18 20:20:13.557 WristbandAssist[551:94993] Upload complete

can you give some suggest,thanks very much~!

mostafaberg, 2016-05-19:
Hi @eirstone!, Are you still experiencing issues with the binary size or has the issue been resolved ?

closed 2016-05-27 by mostafaberg