NordicSemiconductor/IOS-DFU-Library #21: How is the BLE device chosen that receives the firmware update?

larsblumberg, 2016-05-04:
Hi! Thanks a lot for moving the DFU code out of your nRF toolbox app into a separate library!

How can the library be used to choose the right BLE device to send the firmware to? I didn't see any information [in the documentation](https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/master/documentation/README.md)

PS: Why did you choose to maintain 2 repositories for the same DFU library code?

philips77, 2016-05-04:
Hi, you have to pass the correct Peripheral object to the DFUServiceInitiator, together with the Central Manager that was used to scan it. It's the app that responsible for finding the target device.

larsblumberg, 2016-05-04:
Thanks Aleksander, that helps! I have overseen that in the documentation, you did document it.

closed 2016-05-04 by larsblumberg

mostafaberg, 2016-05-08:
@larsblumberg We currently have both repositories cause we use one for the Pods releases and the other for the legacy installation methods, We're still in the process of organizing our codebase and projects, so till the whole process is complete you can expect to see some things like that to happen :)

larsblumberg, 2016-05-09:
@mostafaberg Thanks for clarifying! Do you accept pull requests?

mostafaberg, 2016-05-09:
No problems, Yeah we do accept for sure, any fellow developer PRs are encouraged :)