NordicSemiconductor/IOS-DFU-Library #22: Error using POD

sdespont, 2016-05-08:
The following error occurs when I try to install POD file (pod install):
[!] The dependency `iOSDFULibrary` is not used in any concrete target.

**My Podfile**
use_frameworks!
pod 'iOSDFULibrary'

Any idea?

mostafaberg, 2016-05-08:
Hey!,
This repository is not Pods supported, this is the iOS-DFU-Library , not the Pods repo.

The correct Podfile should be as follows:
`use_frameworks!
    pod 'iOSDFULibrary'
end`

If that dosen't work, please open an issue in the [correct repository](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library) and we can take it from there :+1:

sdespont, 2016-05-08:
OK, thanks for your quick answer. You can close this issue, I have opened a new one in the correct repo.

mostafaberg, 2016-05-09:
For future reference, Issue resolved [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/1), this issue can now be closed @philips77

closed 2016-05-09 by mostafaberg