NordicSemiconductor/IOS-DFU-Library #23: Device disconnected before sending ACK

NSLogSong, 2016-05-18:
@philips77 hi,i have a error when i dfu update complete,but yesterday it performance good,the log like this:
2016-05-18 20:20:12.522 WristbandAssist[551:94993] Progress: 100% (part 1/1). Speed: 110521.844532 bps, Avg speed: 2037.557484 bps
2016-05-18 20:20:12.581 WristbandAssist[551:94993] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100301
2016-05-18 20:20:12.581 WristbandAssist[551:94993] 10: Response (Op Code = 3, Status = 1) received
2016-05-18 20:20:12.582 WristbandAssist[551:94993] 10: Upload completed in 66.63 seconds
2016-05-18 20:20:12.582 WristbandAssist[551:94993] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-05-18 20:20:12.583 WristbandAssist[551:94993] 0: peripheral.writeValue(0x04, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-05-18 20:20:12.584 WristbandAssist[551:94993] Uploading...
2016-05-18 20:20:12.641 WristbandAssist[551:94993] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-05-18 20:20:12.641 WristbandAssist[551:94993] 10: Validate Firmware (Op Code = 4) request sent
2016-05-18 20:20:12.792 WristbandAssist[551:94993] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100401
2016-05-18 20:20:12.793 WristbandAssist[551:94993] 10: Response (Op Code = 4, Status = 1) received
2016-05-18 20:20:12.793 WristbandAssist[551:94993] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-05-18 20:20:12.794 WristbandAssist[551:94993] 0: peripheral.writeValue(0x05, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-05-18 20:20:12.795 WristbandAssist[551:94993] Disconnecting...
2016-05-18 20:20:13.555 WristbandAssist[551:94993] 10: Activate and Reset (Op Code = 5) request sent
2016-05-18 20:20:13.556 WristbandAssist[551:94993] 15: Device disconnected before sending ACK
2016-05-18 20:20:13.556 WristbandAssist[551:94993] 15: Error 14: Unlikely error.
2016-05-18 20:20:13.557 WristbandAssist[551:94993] 0: [Callback] Central Manager did disconnect peripheral
2016-05-18 20:20:13.557 WristbandAssist[551:94993] 5: Disconnected by the remote device
2016-05-18 20:20:13.557 WristbandAssist[551:94993] Upload complete

can you give some suggest,thanks very much~!

philips77, 2016-05-18:
Hi, as you can see this is logged as a warning (15), not an error(20). You may ignore it.
The device got reset and upload was successful. iOS just failed to handle the disconnect properly, but it doesn't matter for the DFU operation.

NSLogSong, 2016-05-18:
thanks very much,but now when i finished the dfu update,my device name is DFUTarg,but yesterday the name is normal,so,i could not decide my device is update successfully,because i decide success on my device's name is performance normally.

philips77, 2016-05-18:
Hi, ios caches the device name, maybe it didn't update yet. Turn the Bluetooth off and on again and try to scan again.

NSLogSong, 2016-05-19:
OK,thanks very much.

closed 2016-05-19 by NSLogSong