NordicSemiconductor/IOS-DFU-Library #24: Timeout after Sending firmware DFU Packet characteristic...

hiniesta, 2016-06-07:
Hello. I'm trying to use the library inside my App. With the nRF-Toolbox in iOS I can upload the firmware without problem. When I try it in my App, after the Upload Proccess ends, and the "Sending firmware DFU Packet characteristic..." I get a time out error code:

2016-06-07 19:08:18.800 Uploading firmware...
2016-06-07 19:08:20.034 Sending firmware DFU Packet characteristic...
2016-06-07 19:10:26.780  [Callback] Central Manager did disconnect peripheral
2016-06-07 19:10:31.325  Disconnected
2016-06-07 19:10:33.970  Error 6: The connection has timed out unexpectedly.
2016-06-07 19:10:33.972  The connection has timed out unexpectedly. (code: 6)

Any idea how to solve this issue?

Thanks!

mostafaberg, 2016-06-08:
Are you using the same CBCentralManager instance that was used for scanning to do the flashing ? if not try that and let me know how it went

hiniesta, 2016-06-08:
Yes, I'm using the same one.

mostafaberg, 2016-06-08:
Hmm that sounds strange, a good pointer is that it works in with nRF-Toolbox so your firmware is ok, has to be a simple mistake somewhere, can I see a code snippet of the part that triggers this issue and how you're starting it ?, if it's possible for me to clone your repo to debug locally that would be even easier, if not possible, then a couple of snippets would do !

hiniesta, 2016-06-08:
Ok, it's not difficult. I'm extending the functionality of a cordova plugin:  https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m

adding this to the class:

```

- (void)updateFirmware:(CDVInvokedUrlCommand*)command {
    //[self scan];
    NSLog(@"updateFirmware");
    //[readCallbacks setObject:[command.callbackId copy] forKey:@"DFU"];
    [self registerObservers];
    NSLog(@"%@",[command.arguments objectAtIndex:0]);
    
    [DFUCallbacks setObject:[command.callbackId copy] forKey:@"estado"];
    [jsonResultado setObject:@"0" forKey:@"estado"];
    [jsonResultado setObject:@"0" forKey:@"proceso"];
    
    NSString *filterUUID = [NSString stringWithString:[[command.arguments objectAtIndex:0] objectAtIndex:2]];
    selectedFileUrl = [NSURL URLWithString:[[command.arguments objectAtIndex:0] objectAtIndex:0]];
    selectedFirmware = nil;
    
    CBPeripheral *selectedPeripheral = [self findPeripheralByUUID:filterUUID];
    
    NSString *fileNameComponent = selectedFileUrl.lastPathComponent;
    NSString *extension = [[fileNameComponent pathExtension] lowercaseString];
    
    if ([extension isEqualToString:@"zip"])
    {
        selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:selectedFileUrl];
        
        if (selectedFirmware && selectedFirmware.fileName)
        {
            NSData *content = [[NSData alloc] initWithContentsOfURL:selectedFileUrl];
            
            initiator = [[DFUServiceInitiator alloc] initWithCentralManager:manager target:selectedPeripheral];
            [initiator withFirmwareFile:selectedFirmware];
            initiator.forceDfu = [[[NSUserDefaults standardUserDefaults] valueForKey:@"dfu_force_dfu"] boolValue];
            initiator.packetReceiptNotificationParameter = [[[NSUserDefaults standardUserDefaults] valueForKey:@"dfu_number_of_packets"] intValue];
            initiator.logger = self;
            initiator.delegate = self;
            initiator.progressDelegate = self;
            
            // initiator.peripheralSelector = ... // the default selector is used
            
            controller = [initiator start];
            
            
            
            
            
            
        }
        else
        {
            selectedFirmware = nil;
            selectedFileUrl = nil;
        }
    }
    
    
    
}

- (void)logWith:(enum LogLevel)level message:(NSString * _Nonnull)message{
    NSLog(@"Log: %@",message);
}
- (void)didStateChangedTo:(enum State)state{
    NSLog(@"en didStateChangedTo %i",state);
    NSString *key = [[NSString alloc] initWithString:@"estado"];
    NSString *DFUCallbackId = [DFUCallbacks objectForKey:key];
    CDVPluginResult *pluginResult = nil;
    NSNumber *val = [NSNumber numberWithInteger:state];
    [jsonResultado setObject:val forKey:@"estado"];
    pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsDictionary:jsonResultado];
    [pluginResult setKeepCallbackAsBool:TRUE];
        [self.commandDelegate sendPluginResult:pluginResult callbackId:DFUCallbackId];
    
    
    
}
- (void)didErrorOccur:(enum DFUError)error withMessage:(NSString * _Nonnull)message{
    NSLog(@"didErrorOccur %@",message);
    NSString *key = [[NSString alloc] initWithString:@"estado"];
    NSString *DFUCallbackId = [DFUCallbacks objectForKey:key];
    CDVPluginResult *pluginResult = nil;
    pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_ERROR messageAsString:message];
    [pluginResult setKeepCallbackAsBool:TRUE];
    [self.commandDelegate sendPluginResult:pluginResult callbackId:DFUCallbackId];
    
    //completado = 6
}
- (void)onUploadProgress:(NSInteger)part totalParts:(NSInteger)totalParts progress:(NSInteger)progress currentSpeedBytesPerSecond:(double)currentSpeedBytesPerSecond avgSpeedBytesPerSecond:(double)avgSpeedBytesPerSecond{
    //NSLog(@"en onUploadProgress %i",progress);
    NSNumber *val = [NSNumber numberWithInteger:progress];
    [jsonResultado setObject:val forKey:@"proceso"];
    
    NSString *key = [[NSString alloc] initWithString:@"estado"];
    NSString *DFUCallbackId = [DFUCallbacks objectForKey:key];
    CDVPluginResult *pluginResult = nil;
    pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsDictionary:jsonResultado];
    [pluginResult setKeepCallbackAsBool:TRUE];
    [self.commandDelegate sendPluginResult:pluginResult callbackId:DFUCallbackId];
    
}

```

hiniesta, 2016-06-15:
Any help?

mostafaberg, 2016-06-15:
Hi @hiniesta sorry for the late response, seems like your issue is happening somewhere else so I'll need to see what happens after the line `2016-06-07 19:08:20.034 Sending firmware DFU Packet characteristic...`

can you log what the response notification data is after doing that call ? or does it disconnect before getting a notification ?

what board version are you using and what kind of firmware ?

hiniesta, 2016-06-15:
The problem is that there is no response, it just disconnected. I'm debugin and I have found something rare. In this execution `self.dfuPacketCharacteristic!.sendNext(number, packetsOf: firmware, andReportProgressTo: progressDelegate)` number = 0. If I check the same with the nRF Toolbox number=12 . Can be that the problem?

mostafaberg, 2016-06-15:
Yes, that is the problem, packet receipt notification needs to be set to 12 packets for it to work properly, if you delete the line where you set it to 0 it will automatically use the default value, which is 12!

let me know how it works out !

hiniesta, 2016-06-15:
Thanks very much! It was that error!!!

mostafaberg, 2016-06-15:
Awesome ! good to know it was a simple fix, sad it took so long to find out, at least you made it!

closed 2016-06-15 by mostafaberg

philips77, 2016-06-15:
In the code you posted you take the value from Settings.
```initiator.packetReceiptNotificationParameter = [[[NSUserDefaults standardUserDefaults] valueForKey:@"dfu_number_of_packets"] intValue];```
 It looks as if the default value was 0, in comparison to 10 or 12 in nRF Toolbox.