NordicSemiconductor/IOS-DFU-Library #25: Error when Dfu_v1 firmware is greater than 0xFFFF bytes

sschiffli, 2016-06-08:
The exact error we are seeing is "Error: 65760 bytes were sent while 224 bytes were reported as received"

The issues stems from the fact that older bootloader versions represented the received firmware image size as a uint16_t (thus we saturate) and sometime later it changed to uint32_t.  This library always interprets the field as a uint32_t.  https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/master/DFULibrary/DFUControlPoint.swift#L196

Any ideas on how to work around this?

philips77, 2016-06-15:
Hi, 
As far as I know the DFU was introduced in SDK 4.3.0. The [documentation](https://developer.nordicsemi.com/nRF51_SDK/nRF51_SDK_v4.x.x/doc/html/group__dfu__ble__service__spec.html) says it was uint32 since the beginning.
If you have even older version that I don't know about, you may just comment out this check in the lib, but you then have to use this library and include it in your workspace manually, not the Pods version.

sschiffli, 2016-06-15:
It looks like there must of been an implementation bug then, because we are using SDK 5.2.0.

If you look at the file dfu_transport_ble.c, line 87, from SDK 5.2.0.39364, path "SDK_v5.2.0.39364\Nordic\nrf51822\Board\nrf6310\device_firmware_updates\bootloader", you will see the variable m_num_of_firmware_bytes_rcvd is defined as a uint16.

If you look at the same file in SDK 6.1.0, its definition is a uint32 as expected, so it got fixed somewhere between 5.2.0 and 6.1.0.

philips77, 2016-06-15:
Well... thanks :) The Android version of DFU library is gently ignoring this value so it should work for you at least with Android. I'll ask SDK guys about it and we may remove this check if there is no DFU Version characteristic.

mostafaberg, 2016-11-30:
Closing this since it's too old and now the library is obsolete, please use the one in the link and repost any issues there if anything is still buggy
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library

closed 2016-11-30 by mostafaberg