NordicSemiconductor/IOS-DFU-Library #27: How to import library in objective-c using cocoapods

hiniesta, 2016-06-20:
This question may have a quick answer, but can't find a solution. I have an objective-c project. I have added firstly the Library buy using workspace, as Method 2. This works all OK.

After that, I have decides to use cocoapods, to import the framework. The problem now is that I can't find what I need to import use the Library in my custom objetive-c class. I have error such as: "Unknow type name DFUServiceController" Any help?

mostafaberg, 2016-06-24:
Hi @hiniesta , you can import swift modules into Objective-C projects with no problems, you need to import the `target_name-Swift.h` auto generated file to make use of the swift code as Obj-C, take a look at how to do that [here](https://developer.apple.com/library/ios/documentation/Swift/Conceptual/BuildingCocoaApps/MixandMatch.html)

FYI: in the future you may post questions that are not issues into our Nordic DevZone to get faster responses

closed 2016-06-24 by mostafaberg

philips77, 2016-06-24:
Are you sure you opened the correct workspace file generated by Pods? And that you have removed all old references to your directly imported DFU project?