NordicSemiconductor/IOS-DFU-Library #30: Swift 3.0 Branch

stcollins, 2016-07-12:
Creating a ticket to see what the status is with a new Swift 3.0 branch and any progress towards compatibility.

mostafaberg, 2016-07-12:
Hi @stcollins, We have still not worked towards Swift 3.0 since it's planned the official release late 2016, we will eventually start working on that branch and fully support it before the final release, but for now we are focused more on getting our codebase running bug free in the current production version of Swift `2.2`, we will push out a swift `3.0` feature branch for developers before finally adding it as an official release before it's officially published

if you have any special questions don't hesitate asking!

clementleroy, 2016-09-14:
Any update on the release now that swift 3.0 is officially here? What about a swift 2.3 branch?

pruthvikar, 2016-09-15:
Hey @clementleroy , I've forked and migrated it to Swift 3.0 over here https://github.com/pruthvikar/IOS-DFU-Library

philips77, 2016-09-15:
Hi, thanks! @mostafaberg is on vacations now, he'll take care of it when back.

mostafaberg, 2016-09-26:
@pruthvikar I'll be resuming work on Swift 3.0 now, the release is going to be within this week

huguesbr, 2016-09-27:
@mostafaberg any plan to also support swift 2.3 (within it's own branch or with in-code version check?)

philips77, 2016-09-27:
We only plan to migrate to 3.0. Th emigration is almost done, couple of days more needed. We are skipping 2.3, afaik not to having to many migrations, which is not healthy.

huguesbr, 2016-09-27:
@philips77 ok too bad. Swift 2.3 is really a step towards Swift 3.0 and the only way to compile on Xcode 8 (beside 3.0) and then for iOS 10. So any app wanting to use your library without migrating to Swift 3.0 and wanting to support iOS 10 will be stuck... I guess I will use a fork then.. thx

mostafaberg, 2016-09-27:
Yeah we're sorry about that, we decided to migrate right to Swift 3.0 to avoid multiple migrations, since Swift 2.3 will eventually be deprecated, so it might be a bit extra work right now, but on the long run you'll be saving time.

once we are done with Swift 3.0 we will re-valuate this decision, and if there's enough time and resources we might be able to do it, but can't promise that ofcourse :)

until then you can use a fork ofcourse and a PR will be more than welcome 👍

huguesbr, 2016-09-28:
@mostafaberg I have a swift 2.3 fork working along with iOS 10 (CBCentralState issue..) but dumping Zip support (faster for me and I don't use it)... https://github.com/wiredbeauty/IOS-Pods-DFU-Library/tree/feature/swift2.3

mostafaberg, 2016-09-28:
Great work, I'll see what can be done when we free up some resources, i'll keep you updated :)

mostafaberg, 2016-09-28:
[Swift 3.0 is now supported](https://cocoapods.org/pods/iOSDFULibrary) in the Cocoapods repo, this repo might be deprecated soon so we will evaluate how urgent it is to migrate it to Swift3.0 within this week or later.

I would recommend going up to Swift 3.0 as soon as you can and avoid any intermediate versions.

closed 2016-09-28 by mostafaberg