NordicSemiconductor/IOS-DFU-Library #33: iOS 10 compatibility

larsblumberg, 2016-07-27:
Hi! We've shipped our app with the nordic iOS library – it works very well for our iOS 9 users. All iOS 10 users however complain that the firmware update process always gets stuck in initialization/starting phase. I haven't yet updated our dev devices to iOS 10. Have you got any feedback for iOS 10 at this point?

mostafaberg, 2016-07-27:
Hi @larsblumberg ! we have not looked into that yet, but will start very soon testing that and will let you know if there are any compatibility issues, if there is we will resolve and have an update out shortly

larsblumberg, 2016-07-27:
Thanks @mostafaberg – I'm looking forward to your testings and feedback in this issue.

mostafaberg, 2016-07-28:
Ok I did some initial testing yesterday and it seems like our library requires minor changes to fully support `iOS10`, the only problem is that our dependencies aren't fully migrated to `Swift-2.3` and `Swift-3` yet, I decided to assign some time to migrate those dependencies and send a PR to the maintainers to speed up the process a bit, of course it'll be up to them to accept or not, but we can still reference the branches I create until they accept the PR or roll out their own fix for the new Swift version.

ETA will be about a working week from Today but, it'll probably be done before that, if you have any questions please feel free to ask here

larsblumberg, 2016-07-28:
Oh dear, I understand that you can't fix these issues with the current release of Xcode 7 and Swift 2.2 because it won't let you run the code on an iOS 10 device. Since we have to use Xcode 7 to release we our app we can't release an update until Apple releases iOS 10 and Xcode 8.

mostafaberg, 2016-07-29:
You are still able to download and use Xcode 8 Beta 3, it also runs side by side with Xcode 7, I'm currently in the process of attempting to remove as much dependencies as possible to avoid future issues like that, currently AVReflection library is removed and we only have the `Zip` pod to work with

larsblumberg, 2016-07-29:
I'm using Xcode 8 besides Xcode 7, but I'm afraid that your fixes for iOS 10 will be available only to Xcode 8 since you mentioned that you will make use of Swift 2.3 and Swift 3.

Do you see a chance that your library supports iOS 10 that we can compile with Xcode 7 in order to submit an app update to the AppStore? Apps built with Xcode 8 can't be submitted as it's still in beta.

mostafaberg, 2016-07-29:
I misunderstood your point, yeah this will not work as you said.. to do that we will have to wait until apple notifies us to start submitting iOS 10 apps, we will get some time to update before iOS 10 is out.

larsblumberg, 2016-07-29:
Can you locate the issue that breaks firmware updates on iOS 10 using Xcode 8 and then back-port that fix to the Swift 2.2 version of this iOS library?

mostafaberg, 2016-08-01:
Yeah i'm actually looking into that, problem with building in Xcode 10 is that lost of the codebase changes, which means the build will never be exactly the same as we have with the current version, so it's kind of hard to back port, i'm trying to get the current production version to fail and attempt to see if there are any logs that are useful, what error does your users exactly report ? any screenshots or error messages ?

mostafaberg, 2016-08-01:
I think I have a fix on the way, will need to do some testing first so i'll say it'll be released tomorrow.
Seems like the problem isn't Swift related, but iOS10, might be a bugfix or a bug, but i'm having the workaround and will let you know soon.

larsblumberg, 2016-08-01:
Thanks for your update! Our users iOS 10 will be happy since we've got 5-10% iOS 10 users, the rest is iOS 9.

What the users actually report is that firmware update is stuck in the "Starting" phase as defined in enum `State` in file `DFUServiceDelegate`. BTW, `State` should probably be prefixed with something (maybe DFUState?) to make this name less generic.

mostafaberg, 2016-08-01:
Good to hear, Beta users always expect bugs you know, yes i managed to reproduce that and i can confirm it's resolved, just need testing for now and will push it out tomorrow

yes it is renamed to DFUState, there is an deprecation attribute that will warn you about that, just depends on which branch you're pulling from, both cases it will all be merged in the near future and most of those naming issues are resolved, thanks for your feedback:)

larsblumberg, 2016-08-01:
Exactly, we're telling our iOS 10 users that they're using a beta version and they're ok with it for now. For now we point them to our Android app and a macOS app where firmware updating works fine with your libraries. Good to hear `State` is now becoming a consistent name.

mostafaberg, 2016-08-02:
Tested and now resolved, this works on both this repo and the cocoapods repo

https://github.com/NordicSemiconductor/IOS-DFU-Library/commit/472060fd32635db1cad57dd0d8dab8c65ab82aca

closed 2016-08-02 by mostafaberg

larsblumberg, 2016-08-02:
Awesome! Thanks a lot!

mostafaberg, 2016-08-02:
No problems, let me know if you encounter any strange behaviours !

larsblumberg, 2016-08-02:
Will do