NordicSemiconductor/IOS-DFU-Library #35: DFUFirmware  is nil， why ？ by setup two or three，selectedFileUrl is exist.

xRuby, 2016-08-09:
DFUFirmware* selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:selectedFileUrl];

mostafaberg, 2016-08-09:
Hi xRuby!, Is the zip file copied over in the build steps ? might be just lying there in Xcode but is not copied to your device when you build, can you print out the output of `selectedFileUrl` ?

xRuby, 2016-08-10:
Yes, it is copied over in the build steps, and can get size of the file, but why the selectedFirmware is nil too.  I think the method is not do as before, but I can't debug it,

mostafaberg, 2016-08-10:
Can you log the value of `selectedFileUrl` and paste it here ?

xRuby, 2016-08-10:
Hi,  thank you very much, I do it  by  first method (pod 'iOSDFULibrary'),  it can upgrade.  i will try to do it by  second and third method later. thank your answer.

mostafaberg, 2016-08-10:
Changing the method won't fix the problem, it's the same library, just different installation methods.

I can't help without getting any more details from you I'm afraid, however it seems like an issue with your app and not the library since it works when we test it

xRuby, 2016-08-10:
Hi, I do it  by second method,  add iOSDFULibrary.framwork  to my project, and it is exist in link binary. but when i  '#import "iOSDFULibrary-swift.h"', the error is file not found. when i '#import <iOSDFULibrary/iOSDFULibrary-Swift.h>' , the selectedFirmware is nil.

xRuby, 2016-08-10:
when i '#import <iOSDFULibrary/iOSDFULibrary-Swift.h>' , the selectedFirmware is nil.

mostafaberg, 2016-08-10:
I'm not asking about the `selectedFirmware`, I'm asking about `selectedFileUrl` please give me that value so I can help you. as I said earlier, the problem is not in the library so installing it in a different way won't help

xRuby, 2016-08-10:
selectedFileUrl is '/var/containers/Bundle/Application/A43CAB95-636A-4F37-B05A-FEE2D2EC0D8D/BodyPlus.app/firmwares/nrf_0005_160810.zip' , and i can get size of the file by the path.

mostafaberg, 2016-08-10:
perfect, does the library show anything special in the logs ?

xRuby, 2016-08-10:
there is nothing...

mostafaberg, 2016-08-10:
That's very strange, the our zip code never returns nil without throwing an exception, what version of Xcode are you using and what kind of device is this happening on ?, also do you see any other logs than from the library at all ?

xRuby, 2016-08-10:
is use `#import <iOSDFULibrary/iOSDFULibrary-Swift.h>` ?   but use `#import "iOSDFULibrary-swift.h"`, the error is the file not found.

xcode: 7.3    device: 6p, 9.3.2

mostafaberg, 2016-08-10:
I would recommend going back to cocoapods to avoid that kind of confusion. I'm not sure which one you are using but the general consensus is `#import target_name-Swift.h`, if it's not found then you need to create the file, again, it's too much of a hassle, just use cocoapods package instead

does your log show anything at all ? if you `NSLog()` something can you see it ? i'm suspecting your logs are hidden for some reason. if you have a screenshot of your Xcode window i might be able to see that for you

xRuby, 2016-08-10:
Another problem, I pod it to upgrade it,  the framework don't do disconnected in DFU and no error delegate do, before upload completed.

mostafaberg, 2016-08-10:
So from what I understand, it starts uploading when you use the cocoapods package ?

closed 2016-08-11 by xRuby

mostafaberg, 2016-08-11:
hi @xRuby did you resolve the problem ?