NordicSemiconductor/IOS-DFU-Library #36: EXC_BAD_INSTRUCTION EXC_I386_INVOP 0x0000000000000000

larsblumberg, 2016-08-16:
We got this crash report when flashing from an OS X app powered by your DFU library, version 0.1.8. Looks like access to a `nil` reference. It's the first time that it's occurring – might be non-relevant:

```
Crashed: com.apple.main-thread
EXC_BAD_INSTRUCTION EXC_I386_INVOP 0x0000000000000000
0	EVReflection	
_TTSf4d_g_d___TZFC12EVReflection12EVReflectionP33_B5E655244C8CE8B4EE02BAF482E9510D14bundleForClassfTPMPs9AnyObject_6bundleCSo8NSBundle_SS + 1202
1	EVReflection	
	_TZFC12EVReflection12EVReflection19setBundleIdentifierfPMPs9AnyObject_T_ + 119
2	iOSDFULibrary	
	DFUStreamZip.swift line 171
	specialized DFUStreamZip.init(urlToZipFile : NSURL, type : UInt8) throws -> DFUStreamZip
3	iOSDFULibrary	
	DFUStreamZip.swift line 0
	DFUStreamZip.__allocating_init(urlToZipFile : NSURL, type : UInt8) throws -> DFUStreamZip
4	iOSDFULibrary	
	DFUFirmware.swift line 115
	DFUFirmware.init(urlToZipFile : NSURL, type : DFUFirmwareType) -> DFUFirmware?
```

mostafaberg, 2016-08-16:
Interesting, is `urlToZipFile` nil in that case ? or another variable ? both cases seems like the `EVReflection` library can't find the manifest file in the Zip, does that exist in the Zip file ?

Also, we have a version that doesn't use `EVReflection` coming out on the way, it's currently available on (develop)[https://github.com/NordicSemiconductor/IOS-DFU-Library/tree/feature/EVReflectionStrip]
you might give that a try if everything else fails.

larsblumberg, 2016-08-16:
Unfortunately this is only a crash report I've got from a user, this said I can't tell you the state of variables. The crash report is the only information I have.

Really looking forward to the solution with reflection – sounds like something more robust!

mostafaberg, 2016-08-17:
Oh it's a crash report, I understand, maybe you can direct the developer to post about it here so we can debug further, otherwise we can't do so much.

Yeah it's going to be definitely more robust without using reflection to parse the manifest file, we can fail gracefully at least in a controlled manner to avoid such situations

larsblumberg, 2016-08-17:
I'm the developer – what additional information do you need?

mostafaberg, 2016-08-17:
That's great, If you can reproduce the bug i'd like to know how it's done, and I would also like to know if you can get hold of the zip file they were trying to flash ?

larsblumberg, 2016-08-17:
We only have one user so far who experiences this problem (however permanently) – we can't reproduce it on ourselves.

Here's however the firmware ZIP file: https://www.senic.com/files/nuimo-firmware-2.2.3.zip

mostafaberg, 2016-11-30:
Closing this since it's too old and now the library is obsolete, please use the one in the link and repost any issues there if anything is still buggy
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library

closed 2016-11-30 by mostafaberg

larsblumberg, 2016-12-01:
@mostafaberg So now you have only 1 repository for the iOS/macOS DFU library? That's good news!

mostafaberg, 2016-12-01:
@larsblumberg yes we do, we are going through a cleanup process and we are trying to maintain a minimal set of apps/libraries that will support all your needs in promise of greater quality and support, this repo will be gone soon too and we will focus forward only on the newer updated library :)

thanks for the support!