NordicSemiconductor/IOS-DFU-Library #38: i'm trying to send next packet but callback return always error

cheignon, 2016-09-05:
in  function  sendNext(number:UInt16, packetsOf firmware:DFUFirmware, andReportProgressTo progressDelegate:DFUProgressDelegate?) 

the sending speed is not managed .

so i replace 
 peripheral.writeValue(packet, forCharacteristic: self.characteristic, type: CBCharacteristicWriteType.WithoutResponse)

by 

  dispatch_after(dispatch_time(DISPATCH_TIME_NOW, Int64(0.1 * dispatch_time_second * Double(NSEC_PER_SEC))), dispatch_get_main_queue(), {
    
           
                peripheral.writeValue(packet, forCharacteristic: self.characteristic, type: CBCharacteristicWriteType.WithoutResponse)

            });

mostafaberg, 2016-09-05:
This is quite wrong and will not work, what are you trying to accomplish ?

cheignon, 2016-09-06:
For me it's works but if you are a best solution, I'm ok for use it.

> Le 5 sept. 2016 à 21:59, Mostafa Torbjørn Berg <notifications@github.com> a écrit :
> 
> This is quite wrong and will not work, what are you trying to accomplish ?

mostafaberg, 2016-09-06:
Can you tell me what the problem you had was that made you need that change ? I can help further if i know what the issue is

cheignon, 2016-09-06:
When i try to update the firmware my Bluetooth board disconnect automatically after 6 packet. So i think the sending speed it's too high.

Cordialement,
Cheignon Dorian

cheignon, 2016-09-06:
> hmmm, which board is it ?
> yes it might be the speed on some boards, what you need to do is set the packet receipt notification number to 1 or 2 and see if that helps, we have that just for this reason

Okay i try with notification = 1

Cordialement,
Cheignon Dorian

mostafaberg, 2016-11-30:
Closing this since it's too old and now the library is obsolete, please use the one in the link and repost any issues there
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library

closed 2016-11-30 by mostafaberg