NordicSemiconductor/IOS-DFU-Library #39: Turning Bluetooth Off During Update. Inconsistent Errors.

morgz, 2016-09-08:
During the `.Uploading` State of a DFU update. If I turn **off** my device's bluetooth a few things happen.

Either:

**Case 1** (This is the most occurrent.)

func logWith(level:LogLevel, message:String):
`
Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: [Callback] Central Manager did disconnect peripheral without error
Lvl LogLevel: message: [Callback] Central Manager did update state to: Powered OFF
Lvl LogLevel: message: Upload aborted`

func didErrorOccur(error:DFUError, withMessage message:String):
`No errors received.`

func didStateChangedTo(state:State):
`No state came through`

**Case 2**

This one is harder to get it to do... 

func didErrorOccur(error:DFUError, withMessage message:String):
`Error occured: DFUError, message: Writing to characteristic failed`

**Case 3**

I don't get an error, but I do get a .Completed state through. Seeing as the update wasn't completed, but only reached about 20% this feels bizarre to me. 

func didStateChangedTo(state:State):
`.Completed`

func didErrorOccur(error:DFUError, withMessage message:String):
`No errors received.`

```
Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: [Callback] Central Manager did disconnect peripheral without error
Lvl LogLevel: message: Disconnected
```

morgz, 2016-09-08:
Worth noting that I'm creating the initial manager and connecting to the CBPeripheral through this library: https://github.com/Polidea/RxBluetoothKit Then accessing the underlying CbCentralManager & CBPeripheral to pass onto the DFUServiceInitiator.

If these cases look completely bizarre to you, I can do a test not using Rx to see if it makes a difference.

mostafaberg, 2016-09-08:
Thanks for the detailed information, this is quite clear and easy to reproduce, will look into it soon and will update you.

until then it's worth noting that this DFU Library is a few versions behind the new release that came out this week, i would recommend giving that a try too as we have lots of bugfixes in there

[iOSDFULibrary Pod](http://www.cocoapods.org/pods/iOSDFULibrary)

if using cocoapods isn't possible for you, there will be an update here in the next 2 weeks

morgz, 2016-09-08:
OK. I was on 1.0.8 - Will try again on 1.0.12

huguesbr, 2016-09-28:
isn't it a dup of https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/16? should close this one, no?

mostafaberg, 2016-10-05:
[This issue was resloved](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library) in our new cocoapods library, I recommend switching to this library instead since we will be deprecating this standalone library

closed 2016-10-05 by mostafaberg