NordicSemiconductor/IOS-DFU-Library #40: Disconnecting Remote Device. Inconsistent behaviour.

morgz, 2016-09-08:
Disconnecting Peripheral during an update by severing BLE connection (shoving it in a crisp packet). Both done during the `.Uploading` state.

**Case 1:**
All good... I get an error through and can react.

`Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: [Callback] Central Manager did disconnect peripheral
Lvl LogLevel: message: Disconnected
Lvl LogLevel: message: Error 6: The connection has timed out unexpectedly.`

`Error occured: DFUError, message: The connection has timed out unexpectedly. (code: 6)`

**Case 2:**
Second Attempt after Case 1, no error comes through.

I don't get an error through, but the logs read:

```
Lvl LogLevel: message: Writing image sizes (0b, 0b, 104480b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
Lvl LogLevel: message: peripheral.writeValue(0x000000000000000020980100, forCharacteristic: 00001532-1212-EFDE-1523-785FEABCD123, type: WithoutResponse)
Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: Start DFU (Op Code = 1, Upload Mode = 4) request sent
Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: Start DFU (Op Code = 1, Upload Mode = 4) request sent
Lvl LogLevel: message: [Callback] Central Manager did disconnect peripheral
Lvl LogLevel: message: Disconnected by the remote device
Lvl LogLevel: message: Disconnecting...
Lvl LogLevel: message: centralManager.cancelPeripheralConnection(peripheral)
```

**Case 3**
Sometimes it seems to just hang, and wait to be reconnected:

Therefore I don't get any errors through, but if I open my crisp packet it will reconnect.

```
[Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Lvl LogLevel: message: [Callback] Central Manager did disconnect peripheral
Lvl LogLevel: message: Disconnected by the remote device
Lvl LogLevel: message: Last upload interrupted. Restarting device, attempts left : 2
Lvl LogLevel: message: Connecting to DfuTarg...
Lvl LogLevel: message: centralManager.connectPeripheral(peripheral, options:nil)
Lvl LogLevel: message: [Callback] Central Manager did connect peripheral
Lvl LogLevel: message: Connected to DfuTarg
Lvl LogLevel: message: Discovering services...
Lvl LogLevel: message: peripheral.discoverServices(nil)
Lvl LogLevel: message: Services discovered
Lvl LogLevel: message: DFU Service found](url)

```

huguesbr, 2016-09-28:
Same issue here... it seem it's not able to reconnect to the device once it switch to DFU mode.
Is this an expected behavior which have to be handle by us (implementing the library) or is it a bug?

mostafaberg, 2016-09-28:
@morgz Sorry for not noticing this earlier, there are two different situations here, if the device restarts in DFU mode it might or might not be broadcasting, if it's broadcasting, you end up getting case 1 and 2, if it's not broadcasting, you'll have to manually reset the peripheral.

we are aware of an issue where it doesn't resume properly when the peripheral is reset during the DFU process, what happens is that the CRC values don't match, my guess is that this happens when the peripheral reports a wrong offset thinking it had received more/less packets than it actually did, causing the CRC calculation to mismatch. the bug we know is that the Library reports a failure instead of restarting the DFU process from scratch. this will be the default behaviour in the fix, which will eliminate the only issue we have control over upon forced disconnection

huguesbr, 2016-09-28:
@mostafaberg great, will attempt to reconnect for now. thx

mostafaberg, 2016-09-28:
@huguesbr Great, keep me updated

huguesbr, 2016-09-28:
@mostafaberg I've build a demo app using swift 2.3, legacy DFU as well as legacy file format (application.bin/application.dat) (yeah I know that's a lot of legacy), but it's demoying the retry attempt mechanism (which work so far..): https://github.com/huguesbr/nrf-ota-demo

mostafaberg, 2016-09-28:
@huguesbr I've taken a quick peek at your project and it doesn't seem so huge to be able to convert it to Swift 3.0, I would suggest doing that first as this will give you the latest version of our library, we can take it from there by opening a new issue on our [CocoaPods repository](https://github.com/NordicSemiconductor/IOS-PODS-DFU-Library) if the problem still persists, also you have lots of `@objc` attributes that are not needed at all, if your code doesn't work without them then something is a bit wrong, I'll try to help you as much as I can to get it up and running, just make sure to do the Swift3.0 conversion and get in touch with me (y)

huguesbr, 2016-09-28:
@mostafaberg sorry I may have mislead you... It's just a demo project (quickly done in one hour today :p). Just to be able to OTA our devices, and also demo it... 

Our project is (way) bigger than that, our code is pretty straightforward to migrate, but we have some dependency on some libraries that haven't migrate (yet).. we're just waiting for Swift 3.0 to settle down a bit :)

About the `@objc` methods, I think it's due to the fact that your protocol are also mark with the `@objc` keyword and my class `OTAManager` is not an obj-c visible class...
ie.: https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/master/DFULibrary/DFUServiceDelegate.swift#L113

mostafaberg, 2016-11-30:
Closing this since it's very old, if there are any more issues please reopen in https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library

closed 2016-11-30 by mostafaberg