NordicSemiconductor/IOS-DFU-Library #48: Xcode 7 Crash on [[DFUFirmware alloc] initWithUrlToZipFile:fileUrl]

rushabhchampaneri, 2016-09-27:
Include the library as a precompiled framework:

When i tried to use XCode7 App crashed on following line:
[[DFUFirmware alloc] initWithUrlToZipFile:fileUrl]

I did check filepath file was exist. Zip file was having json and bin and dat file.

mostafaberg, 2016-09-27:
Hi @rushabhchampaneri !

First thing I recommend using Cocoapods instead of prepcompiled/standalone code

Second thing, do you have a crash log ? at least a call stack would help find out what's the problem. and perhaps a code snippet on what you're doing when it crashes

rushabhchampaneri, 2016-09-27:
@mostafaberg: Ok sure will share those details.

I tried Cocoapods but it EVReflection is not compatible for Swift 3.0 i think.

mostafaberg, 2016-09-27:
@rushabhchampaneri We removed EVReflection on the latest version on the cocoapods repo, and all other dependencies are compatible. but didn't you say you're using Xcode 7 ?

rushabhchampaneri, 2016-09-27:
My bad I have issue with XCode 8. I have added 
use_frameworks!
pod 'iOSDFULibrary'

in pod file and tried to install pod. Let me update issue title

rushabhchampaneri, 2016-09-27:
Incident Identifier: 862F16F8-DF7C-4048-B670-33A2D4420F47
CrashReporter Key:   8e9315ec34760dc5c63fc13be00c3b62d370aa71
Hardware Model:      iPad4,4
Process:             InstallationApp [469]
Path:                /private/var/containers/Bundle/Application/5F7F6781-C215-4919-8882-6E380276366B/InstallationApp.app/InstallationApp
Identifier:          com.enterprise.insigmainc.SmartCoolerAPI
Version:             1 (1.32)
Code Type:           ARM-64 (Native)
Parent Process:      launchd [1]

Date/Time:           2016-09-27 17:40:54.54 +0530
Launch Time:         2016-09-27 17:40:08.08 +0530
OS Version:          iOS 9.3.4 (13G35)
Report Version:      105

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Subtype: KERN_INVALID_ADDRESS at 0x0000000eaeabc028
Triggered by Thread:  0

Filtered syslog:
None found

Global Trace Buffer (reverse chronological seconds):
6.361835     CFNetwork                 	0x00000001831f98e0 TCP Conn 0x1514c5420 SSL Handshake DONE
7.269155     CFNetwork                 	0x00000001831f97f0 TCP Conn 0x1514c5420 starting SSL negotiation
7.270071     CFNetwork                 	0x000000018329be7c TCP Conn 0x1514c5420 complete. fd: 8, err: 0
7.271741     CFNetwork                 	0x000000018329d3a8 TCP Conn 0x1514c5420 event 1. err: 0
38.526929    CFNetwork                 	0x000000018329be7c TCP Conn 0x14ffb7d80 complete. fd: 8, err: 0
38.528961    CFNetwork                 	0x000000018329be7c TCP Conn 0x15107db90 complete. fd: 10, err: 0
38.529390    CFNetwork                 	0x000000018329d3a8 TCP Conn 0x14ffb7d80 event 1. err: 0
38.532076    CFNetwork                 	0x000000018329d3a8 TCP Conn 0x15107db90 event 1. err: 0
39.495572    CFNetwork                 	0x000000018329be7c TCP Conn 0x14ff88240 complete. fd: 4, err: 0
39.497090    CFNetwork                 	0x000000018329d3a8 TCP Conn 0x14ff88240 event 1. err: 0

Thread 0 name:  Dispatch queue: com.apple.main-thread
Thread 0 Crashed:
0   libswiftCore.dylib            	0x0000000100630d64 0x100450000 + 1969508
1   libswiftCore.dylib            	0x0000000100504844 0x100450000 + 739396
2   libswiftCore.dylib            	0x0000000100504760 0x100450000 + 739168
3   libswiftCore.dylib            	0x00000001005073f4 0x100450000 + 750580
4   DFULibrary                    	0x0000000100255510 0x100200000 + 349456
5   DFULibrary                    	0x0000000100231c94 0x100200000 + 203924
6   DFULibrary                    	0x000000010021a768 0x100200000 + 108392
7   DFULibrary                    	0x000000010021a0e0 0x100200000 + 106720
8   DFULibrary                    	0x0000000100211390 0x100200000 + 70544
9   InstallationApp               	0x000000010006d7f4 0x100050000 + 120820
10  InstallationApp               	0x000000010006d488 0x100050000 + 119944
11  Foundation                    	0x00000001835de658 __NSFireDelayedPerform + 428
12  CoreFoundation                	0x0000000182bcd794 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 28
13  CoreFoundation                	0x0000000182bcd438 __CFRunLoopDoTimer + 884
14  CoreFoundation                	0x0000000182bcab4c __CFRunLoopRun + 1520
15  CoreFoundation                	0x0000000182af4c50 CFRunLoopRunSpecific + 384
16  GraphicsServices              	0x00000001843dc088 GSEventRunModal + 180
17  UIKit                         	0x0000000187dd6088 UIApplicationMain + 204
18  InstallationApp               	0x00000001000a5108 0x100050000 + 348424
19  libdyld.dylib                 	0x00000001826928b8 start + 4

Thread 1 name:  Dispatch queue: com.apple.libdispatch-manager
Thread 1:
0   libsystem_kernel.dylib        	0x00000001827b14d8 kevent_qos + 8
1   libdispatch.dylib             	0x00000001826747d8 _dispatch_mgr_invoke + 232
2   libdispatch.dylib             	0x0000000182663648 _dispatch_source_invoke + 0

Thread 2:
0   libsystem_kernel.dylib        	0x00000001827b0b48 __workq_kernreturn + 8
1   libsystem_pthread.dylib       	0x0000000182879530 _pthread_wqthread + 1284
2   libsystem_pthread.dylib       	0x0000000182879020 start_wqthread + 4

Thread 3:
0   libsystem_kernel.dylib        	0x0000000182794fd8 mach_msg_trap + 8
[... 27 lines cut ...]
1   CoreFoundation                	0x0000000182bd31c8 __CFSocketManager + 648
2   libsystem_pthread.dylib       	0x000000018287bb28 _pthread_body + 156
3   libsystem_pthread.dylib       	0x000000018287ba8c _pthread_body + 0
4   libsystem_pthread.dylib       	0x0000000182879028 thread_start + 4

Thread 6:
0   libsystem_kernel.dylib        	0x00000001827b0b48 __workq_kernreturn + 8
1   libsystem_pthread.dylib       	0x0000000182879530 _pthread_wqthread + 1284
2   libsystem_pthread.dylib       	0x0000000182879020 start_wqthread + 4

Thread 7:
0   libsystem_kernel.dylib        	0x00000001827b0b48 __workq_kernreturn + 8
1   libsystem_pthread.dylib       	0x0000000182879530 _pthread_wqthread + 1284
2   libsystem_pthread.dylib       	0x0000000182879020 start_wqthread + 4

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000151544020   x1: 0x0000000100668188   x2: 0x0000000151543b98   x3: 0x0000000000000001
    x4: 0x000000014fe17140   x5: 0x0000000000000020   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000eaeabc008   x9: 0x0000000ffffffff8  x10: 0x00000001046dc1d0  x11: 0x0000000000000018
   x12: 0x0000000000000001  x13: 0x0000000000000082  x14: 0xffffffffffffc000  x15: 0x0000000000003fff
   x16: 0x0000000100622f68  x17: 0x0000000182b13de8  x18: 0x0000000000000000  x19: 0x0000000000000018
   x20: 0x00000001046dc1d0  x21: 0x0000000151543ed0  x22: 0x0000000100668188  x23: 0x0000000151543eb8
   x24: 0x0000000000000004  x25: 0x0000000151543e68  x26: 0x00000001006680a0  x27: 0x0000000000000003
   x28: 0x0000000000000001  fp: 0x000000016fdad270   lr: 0x0000000100504844
    sp: 0x000000016fdad210   pc: 0x0000000100630d64 cpsr: 0x20000000

mostafaberg, 2016-09-27:
Ok that's a different problem then, we have not fully supported Swift 3.0 yet so that's understandable.

you may however (I don't encourage that) to try getting the bleeding edge Swift 3.0 migration branch if you really need it right now, but that's not production ready and contains some bugs, to use that simply update your Podfile to the following:

```
use_frameworks!
target 'MyApp' do
    pod ‘iOSDFULibrary', :git => 'https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library.git', :branch => ‘feature/swift3'
end
```

but as I said, still in development, Secure DFU generally works here, Legacy DFU is still in progress

rushabhchampaneri, 2016-09-27:
Thanks @mostafaberg Just added Crash details. Check if that helps.

mostafaberg, 2016-09-27:
it's not symbolicated so I can't read what's happnening, can you resymbolicate it and update me ?

rushabhchampaneri, 2016-10-25:
Closing this, Works well with Pods.

closed 2016-10-25 by rushabhchampaneri

mostafaberg, 2016-10-25:
awesome news !, great to hear it worked out :)