NordicSemiconductor/IOS-DFU-Library #49: Differences between IOS-Pods-DFU-Library and IOS-DFU-Library repo?

huguesbr, 2016-09-27:
I don't quiet get the difference between the two repos? 
Why do you have two, why not just one with tag for pods?
It seem they not updated at the same time, which one do you plan to maintain most actively?

https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/
https://github.com/NordicSemiconductor/IOS-DFU-Library/

Thanks

mostafaberg, 2016-09-28:
Hi @huguesbr , sorry for the confusion, The `iOS-DFU-Library` is in the process of being deprecated in favor of Cocoapods and SPM, it will be left available for a short period until the IOS-Pods-DFU-Library contains the Swift 3.0 feature, and this will eventually be removed, they are supposed to be 1:1 with no changes but we always apply changes first on this repo before they are reflected on the Standalone library at iOS-DFU-Library repo.

I would say that you can completely ignore the old library and stick with the Cocoapods repository!

closed 2016-09-28 by mostafaberg

huguesbr, 2016-09-28:
@mostafaberg ok, great that's the one I'm on :) thx

mostafaberg, 2016-09-28:
No worries, Swift 3.0 with SPM is coming soon (This week) so everything will be clearer by then

huguesbr, 2016-09-28:
Cool, but SPM is really not well suited for iOS App, so you will keep cocoapods too, right?

mostafaberg, 2016-09-28:
Of course, we're keeping Cocoapods and Carthage (having a little issue with tags on carthage ATM, but it's supported) plus SPM

huguesbr, 2016-09-28:
@mostafaberg 👍