NordicSemiconductor/IOS-DFU-Library #52: What file should import?

LtkDev, 2016-10-07:
I'm import the DFULibrary as source code and do all setup, but my project still can't use library.
I can't call DFUServiceDelegate, DFUServiceInitator...etc.
What should I do to start it ? Is something I miss ?

mostafaberg, 2016-10-07:
Well it surely seems to be a wrong configuration somewhere, We don't recommend importing the source code directly like that, it's better to use Cocoapods, if you can't use cocoapods we do support carthage, this will basically give you a precompiled `.framework` file to import into your project

try going to our [cocoapods page](https://cocoapods.org/pods/iOSDFULibrary) and see the Carthage instructions

LtkDev, 2016-10-07:
@mostafaberg  Thanks a lot.
Hi, I have build iOSDFULibrary.framework by Carthage. And set 
  1.Embed Swift Standard Libraries
  2.New Run Script Phase
  3.New Copy Files Phrase 
  4.Embedded Binaries
  5.Linked Frameworks and Libaries

I got the build failed. Is something I miss?

mostafaberg, 2016-10-07:
I'll need more info on that, what's failing, any messages ?

LtkDev, 2016-10-07:
[image]

mostafaberg, 2016-10-07:
In the top of that class that's throwing the errors can you try to import the generated `ObjC` Header
should be called `iOSDFULibrary-swift.h` and see if that helps, if not you can checkout [this guide](https://developer.apple.com/library/content/documentation/Swift/Conceptual/BuildingCocoaApps/MixandMatch.html) by apple on how to use `Swift` code in `ObjC`

let me know if that helps

LtkDev, 2016-10-07:
@mostafaberg 
It's build succeeded after add this line.
    #import <iOSDFULibrary/iOSDFULibrary-swift.h>
But failed to run as following,
dyld: Library not loaded: @rpath/**Zip.framework/Zip**
  Referenced from: /private/var/mobile/Containers/Bundle/Application/CB4AE11F-EFE1-4379-9AFE-D10781014257/HealthCube.app/Frameworks/iOSDFULibrary.framework/iOSDFULibrary
  Reason: image not found

I'm use clean project to do all step.

mostafaberg, 2016-10-07:
Oh, That would probably be a mistake on our side ! so sorry about that

Can you try adding the Zip library to your cartfile ?
`github "marmelroy/Zip"`

and copy it over to your project same way you did with our library, this should hopefully set things up for you

Joyce306, 2016-10-12:
I have a question that my firmware program is update by about 15% ,than while it shows  a Alert (Upload aborted or Operation not permitted). Could you  tell me  the reason  or how  I can find  the  reason? Thank you very much!  My English is poor and I hope My question is clear. @mostafaberg

LtkDev, 2016-10-13:
@mostafaberg 
Thank you so much! It's worked.

Joyce306, 2016-10-13:
@mostafaberg   I've already done it.  hahaha, I'm  so  happy . Our hardware engineers made a mistake. Now，I got it . 23333

mostafaberg, 2016-10-13:
Great work guys !, i'll have a ticket to relsove this issue within the library so you won't have to deal with dependencies like that !

closed 2016-10-13 by mostafaberg