NordicSemiconductor/IOS-DFU-Library #54: DFU Stuck on Connecting....

sanderdb01, 2016-11-17:
We have a device that has DFU enabled on it, and my iOS code was working, but recently stopped working (around the time of iOS 10 and Swift 3). The latest issue that I have is that I am getting stuck on the didChangeStateTo call back of "connecting". I set the flag high on the DFU service so that I disconnect and change to the bootloader so I am getting DFUTarg and the advertisement. I set the DFUFirmware with a .bin and .dat file from my server. I connect to the DFUTarg peripheral, and once the connection is complete, the following code executes:
let initiator = DFUServiceInitiator(centralManager: self.centralManager!, target: peripheral)
initiator.withFirmwareFile(firmware)
// Optional:
// initiator.forceDfu = true/false; // default false
// initiator.packetReceiptNotificationParameter = N; // default is 12
initiator.logger = self // - to get log info
initiator.delegate = self // - to be informed about current state and errors
initiator.progressDelegate = self // - to show progress bar
// initiator.peripheralSelector = ... // the default selector is used

                    initiator.start()
I then get the call back response of "Connecting" and it gets stuck there. I seem to be implementing the code from nRFToolbox the same way, and as I said before, it seemed to be working fine before.

This is a pretty hot issue for my company, so any help would be greatly appreciated. Thanks a bunch.

philips77, 2016-11-17:
Hi, there will be an update of the library soon (tomorrow or Monday). Could you please test it with the new version then?

sanderdb01, 2016-11-21:
I checked the commits today and I see there was not an update. Has the code been updated so I can test it out? Thanks a bunch.

philips77, 2016-11-21:
Hi, check the Version3 branch.

sanderdb01, 2016-11-21:
will cocoapods install/update install the version 3 branch? If not, how would I go about doing that? Thanks a bunch for the quick response =)

philips77, 2016-11-21:
I thing you can specify the repo and branch in your Podfile. Then, instead of downloading it from Pod, will download from our Github, read here: http://guides.cocoapods.org/using/the-podfile.html

sanderdb01, 2016-11-30:
With the Version3 update, I am now getting a EXC_BAD_ACCESS(code=1,address=0x10) on initiator.with(firmware: firmware). I was not getting this issue before with the master branch. I am printing my details of the firmware beforehand, and that data is there. Also it is a firmware that is already in production and used on Android devices.
I am using a url to an online server, with the .bin and .dat files. Could this be an issue? I tried with the Zip and got errors saying that the zip could not be done (Error while creating ZIP stream: The operation couldn’t be completed. (Zip.ZipError error 0.)). All of the correct files and manifest are there. Thanks again.

mostafaberg, 2016-11-30:
@sanderdb01 Can you please post your issue to the actual library that's having the problem ? this is an obsolete repo 👍, use `https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library`

closed 2016-11-30 by mostafaberg

sanderdb01, 2016-11-30:
I did already, I was posting there just in case. Thank you